cartographer-project/cartographer_ros #640: Can't call

915414922, 2017-12-29:
Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

ojura, 2017-12-30:
You need to `source` the `setup.bash` file, inside the `devel` folder of the workspace where you built Cartographer.

915414922, 2017-12-30:
Thank you very much.I solved it after souce the bash in devel_isolated folder,as I just sourced install_isolated folder before following the document.

closed 2017-12-30 by 915414922