cartographer-project/cartographer_ros #642: loop closure

will1991, 2018-01-04:
This my bag file:  https://drive.google.com/open?id=1pCNcXqBBueuegZfDXI4UqkFQH-A4bDV4
configure file: https://drive.google.com/open?id=1hLFpIKKf2YN2u56-rdTVtLLDmz3HVWwi
luanch file : https://drive.google.com/open?id=1DIxpKiXHWivVhdQ9F348YLm-tDFGzOU1

I want to use odom information to build the map,but I found that it cannot detect the loop closure.
This my result:
[image]

Thank you for your help!!

MichaelGrupp, 2018-01-08:
*"I want to use odom information to build the map"*

What do you mean?

will1991, 2018-01-09:
I build the map  using odometry and laser data

MichaelGrupp, 2018-01-09:
From your lua file:
```lua
...
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry = false,
...
```
You have to set `use_odometry` to true to use both odometry and laser data. See also [my comment here](https://github.com/googlecartographer/cartographer/issues/765#issuecomment-354975118).

gaschler, 2018-01-09:
Generally, for a tuning issue, there may be more effective help when more details are given following the issue template: https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/.github/ISSUE_TEMPLATE

will1991, 2018-01-13:
I do nothing with the main branch, but just add two files ,launch file and configure file. 
I make mistake when I upload the configure file . actually,the configuration is as :
...
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
...

gaschler, 2018-01-26:
Please post the result of our rosbag_validate tool and re-open.

closed 2018-01-26 by gaschler