cartographer-project/cartographer_ros #654: Potential issues with provided odom frame

MichaelGrupp, 2018-01-09:
*This is a follow-up to the meeting we had at Magazino in December (@cschuet, @gaschler, @jihoonl, @mofef).*

@gaschler you mentioned that there are known problems with the provided `odom` frame which can make it an unsuitable choice for (real-time) robotics applications. *If I remember correctly* it was about Cartographer's provided `odom` frame not being guaranteed to conform to [REP 105](http://www.ros.org/reps/rep-0105.html#odom) - correct me if I'm confusing sth. We discussed this because we experienced some instabilities when using Cartographer's `odom` in our robot control instead of pure wheel odometry.

Are there any updates or further information related to this that you can provide? I just remembered this and found no related issue that we can use to track this.

Thanks

gaschler, 2018-01-23:
This is also my understanding.
Currently, I think `odom` is the latest local slam matching result, extrapolated by cartographer_ros.
Ideally, it would be smooth and takes all previous sensor data into account.
(There is no need that it equals local slam results at any point in time, smoothness is important here.)

Thanks for opening this issue.
Have you done measurements or plots yet you can share?

gaschler, 2018-01-29:
This issue depends on #692.

MichaelGrupp, 2018-01-29:
No data because we use the smoother wheel odometry in our robots as it's also used for feedback control. Therefore we encountered instabilities in a test run when replacing it with Cartographer.

To follow your suspicion: I was digging through `node.cc` recently, the relevant section you mentioned is [this](https://github.com/googlecartographer/cartographer_ros/blob/9a63a0479c19d9c267ec9bb62acc58f66d7975dc/cartographer_ros/cartographer_ros/node.cc#L205):
```cpp
    const Rigid3d tracking_to_local = extrapolator.ExtrapolatePose(now);
    const Rigid3d tracking_to_map =
        trajectory_state.local_to_map * tracking_to_local;

    if (trajectory_state.published_to_tracking != nullptr) {
      if (trajectory_state.trajectory_options.provide_odom_frame) {
        std::vector<geometry_msgs::TransformStamped> stamped_transforms;

        stamped_transform.header.frame_id = node_options_.map_frame;
        stamped_transform.child_frame_id =
            trajectory_state.trajectory_options.odom_frame;
        stamped_transform.transform =
            ToGeometryMsgTransform(trajectory_state.local_to_map);
        stamped_transforms.push_back(stamped_transform);

        stamped_transform.header.frame_id =
            trajectory_state.trajectory_options.odom_frame;
        stamped_transform.child_frame_id =
            trajectory_state.trajectory_options.published_frame;
        stamped_transform.transform = ToGeometryMsgTransform(
            tracking_to_local * (*trajectory_state.published_to_tracking));
        stamped_transforms.push_back(stamped_transform);

        tf_broadcaster_.sendTransform(stamped_transforms);
      } else {
        stamped_transform.header.frame_id = node_options_.map_frame;
        stamped_transform.child_frame_id =
            trajectory_state.trajectory_options.published_frame;
        stamped_transform.transform = ToGeometryMsgTransform(
            tracking_to_map * (*trajectory_state.published_to_tracking));
        tf_broadcaster_.sendTransform(stamped_transform);
      }
```
This scribble shows the involved frames (if `provide_odom_frame == true`):
[image]

So essentially we always extrapolate the `odom` to `tracking` pose before publishing with the current version.

gaschler, 2018-02-01:
I did some deeper analysis on the timing of input and odom output, all on the 2D case.
In 2D, when input data is fine (like the Deutsches Museum examples on the download page), the delay from input to LocalSlamCallback is less than 0.02 s. This is very good.
In the current version, `odom` is actually published by a recurring WallTimer job (rather than on each callback) at a period of 0.01 on default.
It uses the most recent LocalSlamResult (and also other input data) and extrapolates with a PoseExtrapolator. So there are up to 0.03 s of extrapolation. Plots look fine, but there is a tiny jerk when a new local Slam result comes in, which is not strictly smooth.
There are a few suspicions I still have.
* If a subscriber controls something based on a numeric derivative of `odom`, this may be jerky.
* The is no notification when local slam is blocked or waiting. In that case, extrapolation just goes on and you will receive unrealistic results.

ojura, 2018-02-02:
> The is no notification when local slam is blocked or waiting. In that case, extrapolation just goes on and you will receive unrealistic results.

Ah yes, I ran into this once. I published a fake clock after the bag ends (just like the offline node does currently if you keep it running, but I had it actually advance). The pose extrapolator just remembers the last observed velocity and keeps integrating it.

cschuet, 2018-02-27:
@MichaelGrupp I think we clarified in our recent meeting that you are not (and probably should not be) using Cartographer's odom topic.

@gaschler  Can we close this issue?

gaschler, 2018-02-28:
The bottom line was that you should only consider the frame `odom` (from output topic `tf`) for real-time robot control when your slowest sensor frequency is ~100 Hz or higher.
You can use it for navigation.
(Closing the issue.)

closed 2018-02-28 by gaschler