cartographer-project/cartographer_ros #655: modified cartographer with a VLP 16 and an imu

zerox3802567, 2018-01-10:
Hi!
I am currently testing cartographer with a vlp16 LIDAR and an imu, so I have some questions and hope you can help me.
I use a VLP-16 and a imu to run cartographer, and i modified four files as attachments. They are
[backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/backpack_3d.launch), [demo_backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_3d.launch) ,[backpack_3d.lua](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua),and [backpack_3d.urdf](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/urdf/backpack_3d.urdf)
After i do 
> roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=${HOME}/Downloads/abc.bag
 I got something error.
[image]

This is my [bag](https://drive.google.com/file/d/1T26jYaoiXBYLWDk29L0PGM58HcEtdrqL/view) file.
Could you help me to check if my codes are wrong?
Thanks a lot!​

gaschler, 2018-01-10:
The roslaunch command syntax seems fine with me.
Perhaps ab**c**.bag?
Closing as not reproducible, hope this helps.

closed 2018-01-10 by gaschler

zerox3802567, 2018-01-10:
So you mean the problem is about the abc.bag file?
Have you already tried the code that were modified and tested it with the bag. file?

gaschler, 2018-01-10:
I was only pointing out the difference between abc in your post and abs in your screenshot.

zerox3802567, 2018-01-10:
Thanks, I got.
I will try it again and tell you the result.

zerox3802567, 2018-01-10:
Hi, I re-input the code `roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=${HOME}/Downloads/abc.bag`
but it still errored
[image]