cartographer-project/cartographer_ros #656: Is Benewake TFMini Lidar module compatible?

kibriashuvo, 2018-01-10:
Hello,

I bought a BW-TFMini LiDAR module. Here is the link of the product.

https://ozrobotics.com/shop/tfmini-lidar-distance-measurement-lidar-sensor-12m-for-drones-and-uav-robots/
I can pan the LIDAR up to 180 degrees with a Pan-Tilt kit along with a typical servo.

However, my question is that does the cartographer algorithm requires a 360 degrees rotating LIDAR for creating 2D maps or 180 degree is ok?

Thank you.

Entropy512, 2018-01-11:
I have had success using Cartographer with a SICK LMS100 that had a good portion of its view obstructed by the vehicle it was mounted to - approximately 200-220 degree FOV.

However, I'm pretty sure that by adding a rangefinder like that to a pan/tilt mechanism, you'll find issues with a combination of:
Spot size too large
Angle repeatability not good enough
Scan rate not high enough

gaschler, 2018-01-12:
@Entropy512 thanks for the helpful answer.
I agree, 180 deg may be not ideal but you can tune it.

Closing as answered.

closed 2018-01-12 by gaschler

kibriashuvo, 2018-01-15:
@Entropy512 Thanks for sharing your experience.

I will try to make something to rotate it 360 degrees and test if it works.