cartographer-project/cartographer_ros #663: gflags related build issue

volkandre, 2018-01-11:
Hi,

I tried to install cartographer_ros as described in the installation instructions, but I get the following error when I run catkin_make:

`==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/dennis/Projects/ws_cartographer/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros'
[1/1] Re-running CMake...
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /home/dennis/Projects/ws_cartographer/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Build type: Release
-- Using CATKIN_DEVEL_PREFIX: /home/dennis/Projects/ws_cartographer/devel_isolated/cartographer_ros
-- Using CMAKE_PREFIX_PATH: /home/dennis/Projects/ws_cartographer/install_isolated;/opt/ros/kinetic
-- This workspace overlays: /home/dennis/Projects/ws_cartographer/install_isolated;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros/test_results
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- looking for PCL_COMMON
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   iostreams
--   regex
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros
ninja: no work to do.
==> ninja -j4 -l4 in '/home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros'
[6/32] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/offline_node.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/dennis/Projects/ws_cartographer/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/local/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/offline_node.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/offline_node.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/offline_node.cc.o -c /home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:35:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_string(
              ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:38:12: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_bool(use_bag_transforms, true,
            ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:40:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_string(pbstream_filename, "",
              ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:42:12: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_bool(keep_running, false,
            ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc: In function ‘void cartographer_ros::RunOfflineNode(std::unique_ptr<cartographer::mapping::MapBuilderInterface>, const cartographer_ros::NodeOptions&, const cartographer_ros::TrajectoryOptions&, const std::vector<std::__cxx11::basic_string<char> >&)’:
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:65:8: error: ‘FLAGS_urdf_filename’ was not declared in this scope
   if (!FLAGS_urdf_filename.empty()) {
        ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:73:8: error: ‘FLAGS_pbstream_filename’ was not declared in this scope
   if (!FLAGS_pbstream_filename.empty()) {
        ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:137:11: error: ‘FLAGS_use_bag_transforms’ was not declared in this scope
       if (FLAGS_use_bag_transforms && msg.isType<tf2_msgs::TFMessage>()) {
           ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:236:7: error: ‘FLAGS_keep_running’ was not declared in this scope
   if (FLAGS_keep_running) {
       ^
[6/32] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/sensor_bridge.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/dennis/Projects/ws_cartographer/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros && /home/dennis/Projects/ws_cartographer/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.
`

Looks like the gflags library cannot be found, but apt says:

`libgflags-dev is already the newest version (2.1.2-3).`

What am I doing wrong?

gaschler, 2018-01-12:
For a build error, consider adding some details as you see fit:
* `locate gflags.h`
* `gcc --version`
* Contents of `(install|devel)_isolated/cartographer(_ros)/devel/CMakeCache.txt`
* `env | grep ROS`
As always, perhaps you already did this, please retry with a current checkout from git and compile again after deleting all build/install folders.

dbergin, 2018-01-19:
1. locate gflags.h =>  located in /usr/include/gflags

2. gcc --version => $ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

3. Contents of (install|devel)_isolated/cartographer(_ros)/devel/CMakeCache.txt
=> I do not see this file in the location specified.
I found the file in build_isolated.  contents below.

I also found the file build_isolated/catkin_make_isolated.cache. # This is the CMakeCache file.
# For build in directory: /home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//The threading library used by boost-thread
BOOST_THREAD_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpthread.so

//build features that require Cartographer gRPC support
BUILD_GRPC:BOOL=OFF

//Build shared libraries (DLLs).
BUILD_SHARED_LIBS:BOOL=OFF

//Boost atomic library (debug)
Boost_ATOMIC_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_atomic.so

//Boost atomic library (release)
Boost_ATOMIC_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_atomic.so

//Boost chrono library (debug)
Boost_CHRONO_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_chrono.so

//Boost chrono library (release)
Boost_CHRONO_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_chrono.so

//Boost date_time library (debug)
Boost_DATE_TIME_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_date_time.so

//Boost date_time library (release)
Boost_DATE_TIME_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_date_time.so

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost iostreams library (debug)
Boost_IOSTREAMS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_iostreams.so

//Boost iostreams library (release)
Boost_IOSTREAMS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_iostreams.so

[... 1339 lines cut ...]
          ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:90:8: error: ‘FLAGS_urdf_filename’ was not declared in this scope
   if (!FLAGS_urdf_filename.empty()) {
        ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:98:8: error: ‘FLAGS_pbstream_filename’ was not declared in this scope
   if (!FLAGS_pbstream_filename.empty()) {
        ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:135:42: error: unable to deduce ‘auto&&’ from ‘bag_filenames’
   for (const std::string& bag_filename : bag_filenames) {
                                          ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:162:11: error: ‘FLAGS_use_bag_transforms’ was not declared in this scope
       if (FLAGS_use_bag_transforms && msg.isType<tf2_msgs::TFMessage>()) {
           ^
/home/dennis/Projects/ws_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/offline_node.cc:261:7: error: ‘FLAGS_keep_running’ was not declared in this scope
   if (FLAGS_keep_running) {
       ^
[7/44] Building CXX object CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/dennis/Projects/ws_cartographer/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/dennis/Projects/ws_cartographer/build_isolated/cartographer_ros && /home/dennis/Projects/ws_cartographer/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

gaschler, 2018-01-23:
Please try adding `#include "gflags/gflags.h"` to offline_node.cc
If this works, I'd be interested why it was ever working. (Maybe through a transitive include?)

Thanks.

gaschler, 2018-01-23:
We just reproduced it and it should be fixed by https://github.com/googlecartographer/cartographer_ros/pull/676
Thanks for reporting.

closed 2018-01-23 by gaschler

dbergin, 2018-01-23:
Yes, this has been my work around. For this file as well as for
offline_node_main.cpp

I was there making the same thing but I could not find the transitive
include.
Dennis

dbergin, 2018-01-23:
Thanks for letting me know.

As I mentioned in the previous email. I also had the issue in
offline_node_main.cpp.
Dennis