cartographer-project/cartographer_ros #665: starting with cartographer - provide imu

PartyMaus, 2018-01-12:
Hello,
I have problems getting cartographer to work. There are no error messages, but the robot frame (base_link) does not move in the map an only rotates in place. I have provided IMU data.

I have one lidar sensor in the front of a car. An IMU in the rear.
Now i try to get cartographer to build me a map. Will it even work in an outdoor environment?

I am trying to run the backpack_2d.launch file. With modified .urdf and .lua files.
This is the link to my modified .urdf and .lua files:
[Folder](https://drive.google.com/drive/folders/1UtqrT-lFSntsZAyykn2MSoEwl8X7-Qit?usp=sharing)

I have run the rosbag_validate and it gives me the following result:
[File](https://gist.github.com/827801890ccd3e7bfc70bce3ac627204.git)

I cannot really say what that tells me. Is my bag okay?

I hope that you could help me. Thank you.

gaschler, 2018-01-12:
Yes, cartographer generally works also in outdoor environments.
rosbag_validate looks fine.
Perhaps you can show a helpful rViz screenshot that explains the behavior?
Sure that no changes to the launch file are necessary (like remapping topics)?
Has cartographer opened a second submap?

PartyMaus, 2018-01-13:
Thank you for your answer. 
I have added a screenshot and provided an update to my launch and configuration files. These should work now. Sorry about that.

Cartographer has not opened a second submap. 
I am thinking i have to tune the parameters to work with my provided sensordata. But I do not know what the parameters actually do.

gaschler, 2018-01-23:
Correct, when you increase TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight or rotation_weight this will make it more costly to move away from the initial estimate before scan matching.
This initial estimate is created by PoseExtrapolator and its rotation depends on IMU data (if there is an IMU). From IMU acceleration, we only use gravity direction.

If you don't see points while moving, the map will be poor in that area.

As with all tuning, please disable global slam `POSE_GRAPH.optimize_every_n_nodes = 0` first. With good data and tuning, local SLAM alone should not drift for multiple submaps.

cschuet, 2018-02-25:
This thread has not seen activity in a while. Please reopen if you have further questions.

closed 2018-02-25 by cschuet

KushnirDmytro, 2019-01-17:
NB: If you are using Gazebo, there are a crucial issue:
Locate where your slam package is  (for me it is with `rospack turtlebot3_slam`)
cd to that directory
edit 'config/turtlebot3_lds_2d.lua': 
on line #23 you will find an ﻿"Easter egg" from developers:
tracking_frame = "imu_link", -- imu_link, If you are using gazebo, use 'base_footprint' (libgazebo_ros_imu's bug)
using that tip: replace "imu_link" with "base_footprint". Without this fix, there will be constant failure mentioning that the IMU frame has to be collocated with your 'tracking frame'.

Fanny-one, 2020-03-17:
Hi, @cschuet @gaschler @davenass @SirVer 
I met similar issue above. I have one lidar sensor and an IMU in the rear which is rotated 90° in yaw axis. However, the robot frame (base_link) does not match with map_frame and odom_frame. The detai is present in [https://github.com/googlecartographer/cartographer_ros/issues/1455](url).
Thanks in advance.