cartographer-project/cartographer_ros #667: testing cartographer with a VLP16 and an imu

zerox3802567, 2018-01-13:
I had a problem in the following connection:
https://github.com/googlecartographer/cartographer_ros/issues/655
The problem still couldn't be solved, and was post in the last comment.
Does anyone have any idea?

ytgcljj, 2018-01-15:
I've tried bag no problem. I can open it normally. You look for other questions.

zerox3802567, 2018-01-15:
So you ran the bag.file with my  launch after modified?

ytgcljj, 2018-01-15:
I use my own here [https://github.com/ytgcljj/cartographer_ros_3d_vlp16](url), You modify demo_vlp16_imu.launch, add

<launch>
  <param name="/use_sim_time" value="true" />

  <include file="$(find cartographer_ros)/launch/vlp16-imu.launch" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_3d.rviz" />
 <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)  " />

</launch>

zerox3802567, 2018-01-15:
hi, I tried the file from your connection, and the result still showed some errors.
[image]
[image]
well, I checked your"vlp16-imu.launch" file, and in line 29 and line 30, I don't think they are right because I just used one LIDAR.  
So do you have any idea about the errors?

ytgcljj, 2018-01-15:
https://github.com/googlecartographer/cartographer_ros/issues/534   You modify  urdf file

ytgcljj, 2018-01-16:
are you OK

zerox3802567, 2018-01-16:
Not yet, there are still some problems.
＃259 I followed from @ozgudum and @SirVer method but my cartographer still didn't work successfully.
I post the problems under the comments  ＃259 as you can see.

zerox3802567, 2018-01-16:
＠ytgcljj 
However, I used your file, and the the terminal showed the error like this
`F0116 20:25:20.442718  3304 configuration_file_resolver.cc:44] File 'trajectory_builder.lua' was not found.
[FATAL] [1516105520.443166403]: F0116 20:25:20.000000  3304 configuration_file_resolver.cc:44] File 'trajectory_builder.lua' was not found.`
Do you have any idea about that?

zerox3802567, 2018-01-16:
hi, @ytgcljj 
I changed the "trajectory_builder.lua" to "trajectory_builder_3d.lua" in the lua file, I thought maybe I can solve the problems, but unfortunately, it appeared another error as the following picture.
[image]
Do you have any idea about that?

ytgcljj, 2018-01-18:
Let me have a look at all your related files.

zerox3802567, 2018-01-18:
Hi, all of the files are totally including three main files.
The first is [ceres-solver](https://github.com/hitcm/ceres-solver-1.11.0.git)
The second one is [cartographer](https://github.com/hitcm/cartographer.git)
And the third one is [cartographer_ros](https://github.com/zerox3802567/cartographer_ros)
All of them were downloaded from the official website connection, only the four files in the third one has been modified.
They were:
[demo_backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_3d.launch)
[backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/backpack_3d.launch)
[backpack_3d.urdf](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/urdf/backpack_3d.urdf)
[backpack_3d.lua](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua)

ytgcljj, 2018-01-18:
The DEM file I gave you was a replay of the bag file, and the other I was debugging

ytgcljj, 2018-01-18:
https://github.com/ytgcljj/ndt_localizer this is bagfile play

zerox3802567, 2018-01-18:
what does the "DEM file" mean?
It seemed that your https://github.com/ytgcljj/ndt_localizer couldn't work, either. The errors were same as the pictures that I post.

ytgcljj, 2018-01-18:
roslaunch cartographer_ros demp_vlp16_imu.launch bag_filename:=${HOME}/XXX.bag
i this run

ytgcljj, 2018-01-18:
In .bag
The frame id of "VLP16" is "velodyne"
The frame id of "imu" is "imu_link"

zerox3802567, 2018-01-18:
so in which file that I should change the frame id?

zerox3802567, 2018-01-18:
You used the configuration files in https://github.com/ytgcljj/ndt_localizer or in https://github.com/ytgcljj/cartographer_ros_3d_vlp16 ? Which files that make your roslaunch successful?

zerox3802567, 2018-01-18:
Hey, these are the configuration files [here](https://github.com/zerox3802567/cartographer_ros/tree/master/cartographer_ros)
I did some changes in the [lua. file](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua) 
Finally, when I input the comment `roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=${HOME}/Downloads/abc.bag` , the result didn't show any error but nothing appeared in the RVIZ. Do you have any idea about that?

zerox3802567, 2018-01-22:
Sorry, I don't understand. You meant you can "roslaunch" successfully?
Which code do you use? Can you test with my bag. file [here](https://drive.google.com/open?id=1T26jYaoiXBYLWDk29L0PGM58HcEtdrqL)
I want to see the result.

zerox3802567, 2018-01-23:
I have seen your connection and read the README, and it had mentioned you run three roslaunch.
Does it mean you do the cartographer in real-time?
Second, I had refer your configuration file and modified mine to become as same as yours.
However, when I do "roslaunch cartographer_ros demo_backpack_3d.launch", the terminal showed the error.
[image]
Do you have any idea about that?

zerox3802567, 2018-01-23:
Moreover, I still have a question about your URDF file, in line 36 and line 56, why did you change the original "velodyne" and "velodyne_joint" into "vertical_vlp16_link" and "vertical_vlp16_link_joint" respectively.
What is the meaning of the modification?

ytgcljj, 2018-01-24:
<launch>
  <param name="/use_sim_time" value="false" />
  <include file="$(find cartographer_ros)/launch/vlp16-imu3d.launch" />
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_3d.rviz" />
</launch>
I don't have a bag_filename to make up a laser, I do the cartographer in real-time
I have a mixture of VLP and IMU in it. Look at it, for example, change it to the same, and see if you can get it up

zerox3802567, 2018-01-24:
Hi, I have totally change my cartographer files as same as yours, but as you can see, the error still appeared.
[image]
What should I do so that I can solve the problems?

ytgcljj, 2018-01-24:
i run roslaunch cartographer_ros demo_vlp16_imu3d.launch  ,you are see  my README.md not demo_backpack_3d.launch

zerox3802567, 2018-01-24:
well, I just didn't change the file name.
Actually,  I changed the files' codes directly.
you can see my configuration:
1. [demo_backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/demo_backpack_3d.launch)
2. [backpack_3d.launch](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/launch/backpack_3d.launch)
3. [backpack_3d.lua](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua)
4. [backpack_3d.urdf](https://github.com/zerox3802567/cartographer_ros/blob/master/cartographer_ros/urdf/backpack_3d.urdf)
You can check these four files which are as same as your files,
1. [demo_vlp16_imu3d.launch](https://github.com/ytgcljj/cartographerVLP16imu3D/blob/master/cartographer_ros/cartographer_ros/launch/demo_vlp16_imu3d.launch)
2. [vlp16-imu3d.launch](https://github.com/ytgcljj/cartographerVLP16imu3D/blob/master/cartographer_ros/cartographer_ros/launch/vlp16-imu3d.launch)
3. [vlp16-imu3d.lua](https://github.com/ytgcljj/cartographerVLP16imu3D/blob/master/cartographer_ros/cartographer_ros/configuration_files/vlp16-imu3d.lua)
4. [vlp16imu3d.urdf](https://github.com/ytgcljj/cartographerVLP16imu3D/blob/master/cartographer_ros/cartographer_ros/urdf/vlp16imu3d.urdf)
The codes of four of my files are totally same yours, so I don't thick it will be the problems.

ytgcljj, 2018-01-25:
You recompile it and run it again

zerox3802567, 2018-01-25:
How do I "recompile" it? What command should I input?

zerox3802567, 2018-01-25:
Hey, I tried `catkin_make_isolated --install --use-ninja` to compile it.
And then I got the error.
[image]

zerox3802567, 2018-01-25:
I added these two code in the backpack_3d.lua file, the result didn't show the error, but unfortunately when I do the rosbag play~ there wasn't any maps appear in the RVIZ. Do you have any idea about that?
`use_nav_sat = false` 
`fixed_frame_pose_sampling_ratio= 1.,`

ytgcljj, 2018-01-25:
If you're just going to debug, then you switch the official DEMO first, and then debug your own

ytgcljj, 2018-01-25:
git clone https://github.com/googlecartographer/cartographer_ros.git 
I can refer to my confecting document：https://github.com/ytgcljj/ndt_localizer

zerox3802567, 2018-01-25:
I actually refer to the official  DEMO and debug it.
I changed my configuration files by refer to your document: https://github.com/ytgcljj/ndt_localizer
After compiling it, I run the `roslaunch cartographer_ros demo_backpack_3d.launch`
And then open the new terminal, `rosbag play ~/Downloads/abc.bag`
Finally, I got nothing in the RVIZ.
So I really have no idea about. If you do the cartograpfer successfully, can you help me test my bag file [here](https://drive.google.com/open?id=1T26jYaoiXBYLWDk29L0PGM58HcEtdrqL)?
Thanks a lot !

ytgcljj, 2018-01-26:
You want to see the bag file, you can
1. Run roscore
2. Run rosbag play ~/Downloads/abc.bag
3, open rviz and change map into your laser's frame ID

zerox3802567, 2018-01-26:
So don’t I need to run the cartographer?
I want to see the the result by testing cartographer with my bag file.

gaschler, 2018-03-29:
Closing for inactivity, please re-open when there is something new.

closed 2018-03-29 by gaschler