cartographer-project/cartographer_ros #668: About the num_laser_scans

MaxChanger, 2018-01-15:
Hello , I have build a 2d map by one laser radar , and I know the default num_laser_scans is 1.
But now , I have two laser radar, and I want to build a 2d map by two  radar,but when I start two node of the laser radar and launch the cartographer ,it's not normal work. One is fixed, and the other is a random move.So I want to know what's wrong and what's mean of the num_laser_scans , it's used to 2d or 3d map?

Thank you !

gaschler, 2018-01-23:
Cartographer supports multiple range sensors.
An example is backpack_3d, which uses two.

MaxChanger, 2018-01-24:
This means that if I use two LIDARs, I'm building a 3D map.
If I want to build a 2d map, advocate the use of a lidar, advocate the use of a laser radar, or to use only one and can not use two or more?

gaschler, 2018-01-24:
No, the number of lasers is independent of whether mapping is done in 2D or 3D.

MaxChanger, 2018-01-24:
But the problem is, when I modify the number of lidar at revo_lds.lua (I write a launch to creat a map by myself and use this lua ), one lidar is fixed, but another TF in rviz of laser radar is moving at random. In other words, if you want to use two lasers, it's not enough to modify the number of lasers in .lua only.

gaschler, 2018-03-29:
Closing for inactivity, please re-open when there is something new.

closed 2018-03-29 by gaschler

bill4u, 2019-09-18:
你好，我用双雷达也出现类似你的问题，请问解决该问题了吗？如何解决的？