cartographer-project/cartographer_ros #670: How to turn back_3d into 2d map

ytgcljj, 2018-01-16:
I have a VLP 16, and I can now see a multi line map of 3d, but there is no way to turn it into 2d map
The configuration files , launch file and urdf file is [here](https://github.com/ytgcljj/ndt_localizer)
Could you help me or give some hint? Thank you!

ytgcljj, 2018-01-16:
bag file is here [](https://drive.google.com/open?id=14zntHzga-wpiCxHX4p8wLZEVdi1TsghL)

ytgcljj, 2018-01-16:
[image]

cschuet, 2018-01-16:
@ytgcljj Are you running Cartographer?

cschuet, 2018-01-16:
[image]

ytgcljj, 2018-01-16:
@cschuet  
yes  But our company's corridors and offices have become a mess map. I don't know where I'm wrong. There's another problem. This is the bag package that can be used. But I want to use laser directly. Why can't you use it? You can help me look at my configuration file. It's wrong.
Thank you for your reply

ytgcljj, 2018-01-16:
[image]

closed 2018-01-23 by ytgcljj