cartographer-project/cartographer_ros #671: The map is not clear, very fuzzy and overlaped

mjason3, 2018-01-20:
Hi, I am trying the cargographer on turtlebot compatible hardware and the output map in RVIZ is quite weird looks like below. Could anyone kindly let me what's wrong in my settings?
[image]

The output of rosbag_validate: https://gist.github.com/mjason3/06159d1abf7c27858f418c83c355dd0e

[Github branch](https://github.com/mjason3/cartographer_turtlebot) for reproduce with command: 
`roslaunch cartographer_turtlebot  play_turtlebot_lidar_2d.launch bag_filename:=/tmp/test_carto_2018-01-20-13-56-40.bag `
Pls. download the recorded bag from my [google drive](https://drive.google.com/file/d/1fp9NTKHOjVwt7Uxv9coTQjDEejROGOjy/view?usp=sharing).

This is the lidar spec i am using: https://github.com/EAIBOT/ydlidar/blob/master/launch/f4.launch

gaschler, 2018-01-23:
I had a quick look at the validation output.

> E0120 14:11:31.667201  8468 rosbag_validate_main.cc:299] Average IMU linear acceleration is 3.92273 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.

This means IMU data will be useless to cartographer. Likely causes are either the measurement format is very strange or you artificially set dropped IMU frames to zero. I believe the latter. Fix this.

> E0120 14:11:31.352073  8468 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)

Please double-check that the odom/ from the bag is neither used by cartographer or rviz.

As with all tuning issues, first resolve input data issues, second tune local slam only (setting `POSE_GRAPH.optimize_every_n_nodes = 0`). Local Slam should drift only after a couple of submaps are created.

Hope this helps.

mjason3, 2018-01-24:
@gaschler yes, it helps which gives me a lot of clues to debug into it. will get back when i get any progress.

gaschler, 2018-01-30:
No problem. Feel free to re-open then.

closed 2018-01-30 by gaschler