cartographer-project/cartographer_ros #678: VLP 3D laser radar storage map operation error

ytgcljj, 2018-01-24:
3D laser radar storage map operation error，
[ INFO] [1516775371.505236371]: I0124 14:29:31.000000  7867 ordered_multi_queue.cc:172] All sensor data for trajectory 1 is available starting at '636523721712195530'.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[cartographer_node-8] process has died [pid 7867, exit code -6, cmd /home/goog/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/goog/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_3d_localization.lua -map_filename /home/goog/data/map3d0.pbstream points2:=vertical_laser_3d __name:=cartographer_node __log:=/home/goog/.ros/log/ee187fce-00cf-11e8-be5c-2c946400477e/cartographer_node-8.log].
log file: /home/goog/.ros/log/ee187fce-00cf-11e8-be5c-2c946400477e/cartographer_node-8*.log

gaschler, 2018-01-26:
Could you please recompile with cmake setting `CMAKE_BUILD_TYPE=RelWithDebInfo`, rerun in a debugger, and post the stack trace?
This would help us tremendously.
We need to see what line the exception was thrown to fix this.
Perhaps this is a bug, perhaps unexpected (empty) input.

ytgcljj, 2018-01-27:
I use a arm system, firefly RTK 3399, I don't know how to input CMAKE_BUILD_TYPE=RelWithDebInfo this life, can you help me to write about it all, is directly under catkin_ws input cmakle CMAKE_BUILD_TYPE=RelWithDebInfo if not help me correct

gaschler, 2018-01-29:
`catkin_make_isolated -DCMAKE_BUILD_TYPE=RelWithDebInfo`

ytgcljj, 2018-02-05:
I'm sorry, I'm wrong. This is the mistake I made when I loaded the local map of 3D.

closed 2018-02-12 by gaschler