cartographer-project/cartographer_ros #679: test Cartographer using Hokuyo UTM-30LX

Vincent1923, 2018-01-24:
Hello!
I have a Hokuyo UTM-30LX LiDAR (2d LiDAR) that publish a sensor_msgs/LaserScan msg. I tested cartographer ROS using the Hokuyo. I have tried with the configuration files which I have modified.
They are demo_hokuyo_2d.launch, hokuyo_2d.launch, hokuyo_2d.lua, hokuyo_2d.urdf. But the result is not very well. I couldn't see the map in the rviz. 

[The configuration file , launch files and urdf file is here](https://gist.github.com/Vincent1923/02bda6247bcf7cff6f0cbea9d462394a).

and [the bag file is here](https://pan.baidu.com/s/1dGJwJXv).

Could you help me or give some hint? Thank you!

gaschler, 2018-01-26:
Please post also the output of the bag file validate tool.

Vincent1923, 2018-01-26:
This is the output of the bag file validate tool:

rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ${HOME}/Downloads/2018-01-22-10-58-19.bag
E0126 17:20:01.859179 125663 rosbag_validate_main.cc:286] frame_id "laser_link" on topic /scan has serialization time 1516687455.978677140 but sensor time 1516589901.955344553 differing by -97554 s.
E0126 17:20:01.859567 125663 rosbag_validate_main.cc:286] frame_id "odom" on topic /odom has serialization time 1516687455.978922450 but sensor time 1516589901.979068671 differing by -97554 s.
E0126 17:20:01.859587 125663 rosbag_validate_main.cc:286] frame_id "odom" on topic /odom has serialization time 1516687455.978947107 but sensor time 1516589901.979113530 differing by -97554 s.
E0126 17:20:01.976425 125663 rosbag_validate_main.cc:316] Point data (frame_id: "laser_link") has a large gap, largest is 0.775893 s, recommended is [0.0005, 0.05] s with no jitter.
I0126 17:20:01.976646 125663 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser_link"):
Count: 2972  Min: 0.017354  Max: 0.775893  Mean: 0.051978
[0.017354, 0.093208)	   #################	Count: 2575 (86.641991%)	Total: 2575 (86.641991%)
[0.093208, 0.169062)	                   #	Count: 217 (7.301480%)	Total: 2792 (93.943474%)
[0.169062, 0.244916)	                   #	Count: 95 (3.196501%)	Total: 2887 (97.139977%)
[0.244916, 0.320770)	                    	Count: 42 (1.413190%)	Total: 2929 (98.553162%)
[0.320770, 0.396623)	                    	Count: 13 (0.437416%)	Total: 2942 (98.990578%)
[0.396623, 0.472477)	                    	Count: 14 (0.471063%)	Total: 2956 (99.461639%)
[0.472477, 0.548331)	                    	Count: 9 (0.302826%)	Total: 2965 (99.764465%)
[0.548331, 0.624185)	                    	Count: 2 (0.067295%)	Total: 2967 (99.831764%)
[0.624185, 0.700039)	                    	Count: 2 (0.067295%)	Total: 2969 (99.899055%)
[0.700039, 0.775893]	                    	Count: 3 (0.100942%)	Total: 2972 (100.000000%)
I0126 17:20:01.976831 125663 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
Count: 3735  Min: 0.000006  Max: 1.218777  Mean: 0.041313
[0.000006, 0.121883)	  ##################	Count: 3397 (90.950470%)	Total: 3397 (90.950470%)
[0.121883, 0.243760)	                   #	Count: 143 (3.828648%)	Total: 3540 (94.779114%)
[0.243760, 0.365637)	                    	Count: 89 (2.382865%)	Total: 3629 (97.161980%)
[0.365637, 0.487514)	                    	Count: 39 (1.044177%)	Total: 3668 (98.206161%)
[0.487514, 0.609391)	                    	Count: 48 (1.285141%)	Total: 3716 (99.491295%)
[0.609391, 0.731269)	                    	Count: 6 (0.160643%)	Total: 3722 (99.651939%)
[0.731269, 0.853146)	                    	Count: 6 (0.160643%)	Total: 3728 (99.812584%)
[0.853146, 0.975023)	                    	Count: 5 (0.133869%)	Total: 3733 (99.946449%)
[0.975023, 1.096900)	                    	Count: 1 (0.026774%)	Total: 3734 (99.973228%)
[1.096900, 1.218777]	                    	Count: 1 (0.026774%)	Total: 3735 (100.000000%)

gaschler, 2018-01-26:
As far as I know, differences between serialization timestamp and sensor timestamp are an error because ROS will extrapolate by the clock (=serialization timestamp). Cartographer looks only at the sensor timestamps, but tf will be off. Please fix the timestamp mismatch in your input data and re-open if needed.

Hope this helps.

closed 2018-01-26 by gaschler

Vincent1923, 2018-01-26:
I think this is because when I recorded the rosbag, there were too many other topics that I didn't want. I played the rosbag again and recorded just two topics, which were "/scan" and "/odom". Is this the reason that causes the differences between serialization timestamp and sensor timestamp.

gaschler, 2018-01-26:
I would image that `rosbag filter` leaves both timestamps unchanges http://wiki.ros.org/rosbag/Tutorials/Producing%20filtered%20bag%20files 

Mismatched timestamps are not supported, therefore our validation check.