cartographer-project/cartographer_ros #684: VLP-16 3D map of underground garage

meng216073, 2018-01-25:
hi, I need help.
when I use cartographer_ros and VLD-16 to build map of underground garage, I can not get the map like the demo_backpack3d.
I use two LIDAR, they are arranged on the front and back of my car. I drove three laps in the underground garage. The last lap passes through the middle of the garage. And I record the data.
Then I run the code:
roslaunch cartographer_ros demo_vlp16-imu-2lidar.launch bag_filename:=/home/gm/2018-1-15-17-24-cartographer-subway.bag
The following is the result I got.
[image]
[image]
[image]

As you can see , the maps are not in the same plane. Can you tell me why? Closed loop detection can not be used in maps.
when I run the code
 `rosrun cartographer_ros cartographer_rosbag_validate -bag_filename /home/gm/2018-1-15-17-24-cartographer-subway.bag`
I get the following results(Although Odom data was collected, it was not used)：
gm@CJ:~/catkin_vlp$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename /home/gm/2018-1-15-17-24-cartographer-subway.bag 
E0125 19:44:39.796569  3960 rosbag_validate_main.cc:121] frame_id utm time 1516009772392129968: Odometry pose is invalid. pose position: 
  x: 0
  y: 0
  z: 0
orientation: 
  x: 0
  y: 0
  z: 0
  w: 0
E0125 19:44:39.796896  3960 rosbag_validate_main.cc:121] frame_id utm time 1516009772402190993: Odometry pose is invalid. pose position: 
  x: 0
  y: 0
  z: 0
orientation: 
  x: 0
  y: 0
  z: 0
  w: 0
E0125 19:44:39.796934  3960 rosbag_validate_main.cc:121] frame_id utm time 1516009772412081307: Odometry pose is invalid. pose position: 
  x: 0
  y: 0
  z: 0
orientation: 
  x: 0
  y: 0
  z: 0
  w: 0
 (the core has been dumped)
gm@CJ:~/catkin_vlp$ 
This is my configuration:
[demo_vlp16-imu-2lidar.launch](https://github.com/meng216073/cartographer_ros/blob/master/cartographer_ros/launch/demo_vlp16-imu-2lidar.launch)
[vlp16-imu-2lidar.launch](https://github.com/meng216073/cartographer_ros/blob/master/cartographer_ros/launch/vlp16-imu-2lidar.launch)
[vlp16-imu-2lidar.lua](https://github.com/meng216073/cartographer_ros/blob/master/cartographer_ros/configuration_files/vlp16-imu-2lidar.lua)
[vlp16-imu-2lidar.urdf](https://github.com/meng216073/cartographer_ros/blob/master/cartographer_ros/urdf/vlp16-imu-2lidar.urdf)
My bag file is [here](https://drive.google.com/open?id=1Mt23ZOYOSODu6Vhi54JBHm28xetmRM1C)
As you can see, my map is very bad. So I have 3 questions:
1.Why are the maps generated not in the same plane？Is it due to the inaccurate position calibration of the two laser radar?  Do I need to calibrate the relative position of the lidar accurately?
2.Why has the generated map deflected in one direction? Is it because of the problem that I configure, or because of the sensor data is not good enough?
3.Why did the generated maps have no closed loop detection?  Or why does closed loop detection have not solved the problem of faults and deflection in the above problems?  Because I think the closed-loop detection may be able to solve the above problems.
By the way,what changes do I need to make if I want to generate 3D point cloud map?
Looking forward to your reply, thank you very very much!

gaschler, 2018-01-26:
> Do I need to calibrate the relative position of the lidar accurately?

Yes, these transforms must the accurate, especially with respect to the IMU.

> Why has the generated map deflected in one direction?

Wrong urdf transforms could explain this. But it could also be something else.

> Why did the generated maps have no closed loop detection? Or why does closed loop detection have not solved the problem of faults and deflection in the above problems? Because I think the closed-loop detection may be able to solve the above problems.

Your submaps generated by local SLAM seem not very good.

Please first correct any sensor input and config issues, then, second, tune local slam alone (turning off optimization).

meng216073, 2018-01-28:
thank you for you reply, let me try . thank you very much.

meng216073, 2018-01-29:
@gaschler Our LIDAR is fixed on the car. If i drive to fast ,the sensor will get little data per circle.  Does it affect the generation of my Local SLAM?

meng216073, 2018-01-29:
@gaschler The scene of my map is a large plane outside, but because of the height difference,The generated map is not on the same plane, and the generation map will disappear automatically. How do I configure it to avoid this problem?
Because I found the demo of the backpack_3d is a two layer Museum, and my experimental scene is a horizontal plane.
And if I want to generate a 3 - dimensional point cloud, how do I need to configure it？ (because the demo of backpack_3d is a map of 2 dimensions )

meng216073, 2018-01-30:
@SirVer I finally found a question similar to me #424 . So， Is the map disappearing because of the inaccuracy of the calibration of the relative position of IMU and laser radar? But， Why is there always a height difference in a certain position? In this location map is divided into two layers, one layer will disappear. Can you tell me why? I have been plagued by this problem for a long time。 Looking forward to your reply! Thank you ~~~

ShineSong, 2018-02-27:
@meng216073 Hi, how about your transformated imu reading, is the Z-value around -G or G?

zdp1994, 2019-12-02:
Hello, I see that you have used two vlp-16 lidar, do you have to use two, or is the effect of two better?

bochen87, 2020-09-11:
closing as long inactivity for tuning related issue

closed 2020-09-11 by bochen87