cartographer-project/cartographer_ros #687: Install cartographer for the following error？

ytgcljj, 2018-01-26:
Install cartographer for the following error, please help me to analyze it, I installed proto3

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/firefly/catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/firefly/catkin_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j6 -l6 in '/home/firefly/catkin_ws/build_isolated/cartographer/install'
[6/279] Running C++ protocol buffer co...correlative_scan_matcher_options.proto
FAILED: cd /home/firefly/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/firefly/catkin_ws/build_isolated/cartographer/install -I /home/firefly/catkin_ws/src/cartographer /home/firefly/catkin_ws/src/cartographer/cartographer/mapping_2d/scan_matching/proto/fast_correlative_scan_matcher_options.proto
cartographer/mapping_2d/scan_matching/proto/fast_correlative_scan_matcher_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[6/279] Running C++ protocol buffer co...mapping/proto/pose_graph_options.proto
FAILED: cd /home/firefly/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/firefly/catkin_ws/build_isolated/cartographer/install -I /home/firefly/catkin_ws/src/cartographer /home/firefly/catkin_ws/src/cartographer/cartographer/mapping/proto/pose_graph_options.proto
cartographer/mapping/proto/pose_graph_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[6/279] Running C++ protocol buffer co...pping/proto/submap_visualization.proto
FAILED: cd /home/firefly/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/firefly/catkin_ws/build_isolated/cartographer/install -I /home/firefly/catkin_ws/src/cartographer /home/firefly/catkin_ws/src/cartographer/cartographer/mapping/proto/submap_visualization.proto
cartographer/mapping/proto/submap_visualization.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[6/279] Running C++ protocol buffer co...ommon/proto/ceres_solver_options.proto
FAILED: cd /home/firefly/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/firefly/catkin_ws/build_isolated/cartographer/install -I /home/firefly/catkin_ws/src/cartographer /home/firefly/catkin_ws/src/cartographer/cartographer/common/proto/ceres_solver_options.proto
cartographer/common/proto/ceres_solver_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[6/279] Running C++ protocol buffer co...pher/mapping/proto/serialization.proto
FAILED: cd /home/firefly/catkin_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/firefly/catkin_ws/build_isolated/cartographer/install -I /home/firefly/catkin_ws/src/cartographer /home/firefly/catkin_ws/src/cartographer/cartographer/mapping/proto/serialization.proto
cartographer/mapping/proto/serialization.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[6/279] cd /home/firefly/catkin_ws/bui...d/cartographer/install/AllFiles.cmake"
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/firefly/catkin_ws/install_isolated/env.sh', 'ninja', '-j6', '-l6']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/firefly/catkin_ws/build_isolated/cartographer && /home/firefly/catkin_ws/install_isolated/env.sh ninja -j6 -l6

gaschler, 2018-01-26:
For a build issue, consider adding some details as you see fit:
* **`protoc --version`**
* If it is related to a dependency, which version and where is it installed?
* `gcc --version`
* Contents of `(install|devel)_isolated/cartographer(_ros)/devel/CMakeCache.txt`
* `env | grep ROS`
As always, perhaps you already did this, please retry with a current checkout from git and compile again after deleting all build/install folders.

closed 2018-01-26 by gaschler

ytgcljj, 2018-01-27:
firefly@firefly:~$ env|grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/opt/ros/kinetic/share
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
ROS_DISTRO=kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros
firefly@firefly:~$ protoc --version
libprotoc 3.4.0
firefly@firefly:~$ gcc --version
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

firefly@firefly:~$ uname -ar
Linux firefly 4.4.77 #554 SMP Thu Nov 30 11:30:11 HKT 2017 aarch64 aarch64 aarch64 GNU/Linux

gaschler, 2018-01-31:
You have the protobuf compiler 3.4 in your system.
Make sure cmake variable `${PROTOBUF_PROTOC_EXECUTABLE}` points to the path of that executable, i.e. `which protobuf`.