cartographer-project/cartographer_ros #690: How do I tell if cartographer_node has finished optimising the trajectory?

Oluwatoni, 2018-01-26:
The /finish_trajectory service doesn't send a response so how can I detect when the optimisations are complete? This is problem especially when working on very large maps when it can take minutes to optimise. Thank you!

MichaelGrupp, 2018-01-29:
The missing response and missing log messages have been annoying me too a few times ;) The `/finish_trajectory` and `/write_state` services should send boolean responses.

I had a look now and the fix was straight-forward - trying to get this into master in this PR: https://github.com/googlecartographer/cartographer_ros/pull/696

ojura, 2018-01-29:
Doesn't `FinishTrajectoryUnderLock` always return true? I don't think this is what @Oluwatoni is looking for. I think for that you would need a service like `get_worker_queue_size` or something similar.

MichaelGrupp, 2018-01-29:
https://github.com/googlecartographer/cartographer_ros/blob/master/docs/source/ros_api.rst

`finish_trajectory` *Finishes the given trajectory_id's trajectory by running a final optimization*

Doesn't `HandleFinishTrajectory` cover all that?

ojura, 2018-01-29:
It does, but it doesn't block until optimization is finished. If you follow what functions are called, you'll get to [`FinishTrajectory` in `PoseGraph`](https://github.com/googlecartographer/cartographer/blob/a7ed7e224f98b396762c865b81b62dc3abea2e81/cartographer/mapping_2d/pose_graph.cc#L398). There you can see that optimization is only _dispatched_ i.e. put onto the worker queue. The function call itself can return immediately, and you get a `true` returned immediately, while the optimization itself can happen later, i.e. when some other thread reaches that point in the worker queue.

MichaelGrupp, 2018-01-29:
True.

MichaelGrupp, 2018-01-30:
Added a comment in my PR to clarify what the response means. Thanks for pointing it out, didn't notice that.

MichaelGrupp, 2020-10-16:
One thing that could be checked now to get a hint of the state of optimization is this metric: https://github.com/cartographer-project/cartographer/commit/1cae11593d813df7be11e45dc3722caadff2899f#diff-03a62bc60e03586a8b9bf5ca74a0cf291728593a2f9905dd2e46cac6e00a59f4

Closing for inactivity.

closed 2020-10-16 by MichaelGrupp

archererer, 2020-11-18:
Is there a way to detect optimization completion programmatically? Such that `WriteState` service call be called automatically once the optimization is done following the `FinishTrajectory` call.

imstevenpmwyca, 2021-03-23:
> One thing that could be checked now to get a hint of the state of optimization is this metric: [cartographer-project/cartographer@1cae115#diff-03a62bc60e03586a8b9bf5ca74a0cf291728593a2f9905dd2e46cac6e00a59f4](https://github.com/cartographer-project/cartographer/commit/1cae11593d813df7be11e45dc3722caadff2899f#diff-03a62bc60e03586a8b9bf5ca74a0cf291728593a2f9905dd2e46cac6e00a59f4)
> 
> Closing for inactivity.

Hello @MichaelGrupp, I have a question regarding this metric.

As in the original issue, I am trying to tell programmatically if the cartographer_node has finished processing all the measurements, performing the optimizations of the posegraph and trajectories, and trimming the submaps (when running the node with the submap trimmer). This last one is particularly interesting because I have noticed that the trimmer could lag way behind to the finish_trajectory service call, especially in large maps. All of this because I want to make sure that the occupancy map and the pbstream saved (by the write_state service) are optimal.

Given the name of the metric you mentioned, "mapping_2d_pose_graph_work_queue_size -> Number of items in the work queue", I am inclined to think that this metric only tells if the cartographer_node has finished processing the measurements, but it gives no indication for the optimizations or trimmer.
Could you confirm if this is right? Or does this metrics indicate the state of the optimization and trimmer?

Thank you very much for your time and consideration!