cartographer-project/cartographer_ros #691: Client does not wait until server is alive

gaschler, 2018-01-29:
When configuring a distributed system with a launch file,
the dependency that the server needs to start up first cannot be formulated.

A work-around is to an agent (which uses a bridge) delayed, so it usually starts after the uplink server.
```
--- a/cartographer_ros/cartographer_ros/cartographer_grpc/node_grpc_main.cc
+++ b/cartographer_ros/cartographer_ros/cartographer_grpc/node_grpc_main.cc
@@ -87,6 +87,7 @@ int main(int argc, char** argv) {
   CHECK(!FLAGS_configuration_basename.empty())
       << "-configuration_basename is missing.";
+  sleep(1);
   ::ros::init(argc, argv, "cartographer_grpc_node");
   ::ros::start();
```

One could call this a launch file bug.
Or, a missing feature to let our client (stubs) wait until server is alive.

dirtshell, 2018-01-31:
I experienced a similar problem. When I tried to launch cartographer, my IMU drivers, and start my lidar at the same time using a launch file, cartographer would fail to start (`map` topic was never published) and no errors would be printed out. Just a blank rviz screen. This may have been because my lidar drivers would start and publish invalid point clouds while the lidar started spinning up.

The workaround I used was by adding a delay to the lidar's start up in my `.launch` file. 

**`.launch`**
```
<launch>

  <!-- IMU ROS Drivers -->
  <node name="imu_driver" 
        pkg="imu_driver"
        type="imu" args="/dev/ttyUSB0 115200" />
  
  <!-- lidar Sensor ROS Drivers -->
  <arg name="host" default="192.168.1.78" />
  <arg name="ns" default="Sensor" />
  <arg name="lidar_client_start_delay" default="5" /> <!-- THIS IS THE IMPORTANT PART -->

  <node name="delayed_lidar_client_launch"
        pkg="cartographer_ros"
        type="timed_roslaunch.sh"
        args="$(arg client_start_delay) lidar_client_ros client.launch host:=$(arg host)"
        required="true"
        output="screen"/>

  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/my_urdf.urdf" />
  
  <node name="robot_state_publisher" pkg="robot_state_publisher"
      type="robot_state_publisher" />
  
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename lidar_2d.lua"
      output="screen">
    <remap from="points2" to="/Sensor/points" />
    <remap from="imu" to="/my_imu/Imu" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/my_rviz.rviz" />
</launch>
```
**`timed_roslaunch.sh`**
```
#!/bin/bash          
#
# Script to delay the launch of a roslaunch file
# 
# Koen Lekkerkerker
# Thu 24 Apr 2014 
#
# Use: ./timed_roslaunch.sh [number of seconds to delay] [rospkg] [roslaunch file]
#

function showHelp(){
    echo 
    echo "This script can delay the launch of a roslaunch file"
    echo "Place it in the 'scripts' folder of your catkin package"
    echo "and make sure that the file is executable (chmod +x timed_roslaunch.sh)"
    echo 
    echo "Run it from command line:"
    echo 
    echo "Use: ./timed_roslaunch.sh [number of seconds to delay] [rospkg] [roslaunch file] [arguments (optional)]"
    echo "Or: rosrun [yourpackage] time_roslaunch.sh [number of seconds to delay] [rospkg] [roslaunch file] [arguments (optional)]"
    echo "Example: ./timed_roslaunch.sh 2 turtlebot_navigation amcl_demo.launch initial_pose_x:=17.0 initial_pose_y:=17.0"
    echo 
    echo "Or run it from another roslaunch file:"
    echo 
    echo '<launch>'
    echo '  <arg name="initial_pose_y" default="17.0" />'
    echo '  <node pkg="semantic_turtle_test" type="timed_roslaunch.sh"'
    echo '    args="2 turtlebot_navigation amcl_demo.launch initial_pose_x:=17.0 initial_pose_y:=$(arg initial_pose_y)"'
    echo '    name="timed_roslaunch" output="screen">'
    echo '  </node>'
    echo '</launch>'
}

if [ "$1" = "-h" ]; then
    showHelp
else 
    echo "start wait for $1 seconds"
    sleep $1
    echo "end wait for $1 seconds"
    shift
        echo "now running 'roslaunch $@'"
    roslaunch $@
fi
```

cschuet, 2018-02-25:
Bridge now has a retry policy for the `AddTrajectory` call. This removes the race condition.

closed 2018-02-25 by cschuet