cartographer-project/cartographer_ros #692: Published transforms are interrupted

gaschler, 2018-01-29:
Expected behavior:
Node publishes tf topic every ~10 ms.

Observes behavoir:
Published tf topic is sometimes interrupted for ~100 ms.

How to reproduce:
Execute the following launch file and visualize the timestamps of the generated bag file.
```
<launch>
  <param name="/use_sim_time" value="true" />
  <include file="$(find cartographer_ros)/launch/backpack_2d.launch" />
  <node name="rviz" pkg="rviz" type="rviz"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      required="true"
      args="--clock $(env HOME)/Downloads/b2-2016-04-27-12-31-41.bag
          --duration=20" />
  <node name="record_tf" pkg="rosbag" type="record"
      required="true"
      args="--output-name=$(env HOME)/Downloads/tf-reproduce.bag
          /tf" />
</launch>
```

```
rqt_bag ~/Downloads/tf-reproduce.bag
```

[image]

MichaelGrupp, 2018-01-29:
FYI: Could reproduce on my machine.

[image]

Just a wild guess: maybe a threading / lock issue? Some optimization / other background task?

gaschler, 2018-01-30:
I did a lot of debugging and my conclusion is this is *intended behavior* of `rosbag play --clock`.
With respect to wall clock time, the largest gap between two `Node::PublishTrajectoryStates` was only 0.020 s, so we publish "odom" in time.

However, the simulated time topic "clock" published by `rosbag play` has exactly these large jumps.
The last message before such jump publishes a simulated timestamp in the future, and then nodes will work in this simulated time, including `rosbag record`.

[image]

Long story short, this does not indicate anything related to real hardware, it only shows we cannot use rosbag (like this) to reproduce timing behavior.

Closing as working as intended.

closed 2018-01-30 by gaschler

ojura, 2018-01-30:
In that case, the offline node is probably affected by this as well. It behaves similarly to rosbag play, and the stamps used there for publishing the clock are serialization timestamps, and _not_ header timestamps (i.e. sensor measurement timestamps):
https://github.com/googlecartographer/cartographer_ros/blob/2538ac3e45ccaf553e956bbb7e745c26008460bf/cartographer_ros/cartographer_ros/offline_node.cc#L279-L280

So, if your bag recording is jumpy (so not the sensor measurements' header stamps, which can be fine, but the serialization times), you get jumpy behaviour when using `sim_time = true`.

gaschler, 2018-01-30:
The input bag [b2-2016-04-27-12-31-41.bag](https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b2-2016-04-27-12-31-41.bag) is not jumpy (neither serialization nor headers).

Thanks for pointing out that offline_node will be similar in this case.