cartographer-project/cartographer_ros #693: Offline node broken for same robot, multiple runs

ojura, 2018-01-29:
@gaschler there is an issue with the topic remapping [recently introduced](https://github.com/googlecartographer/cartographer_ros/blob/9a63a0479c19d9c267ec9bb62acc58f66d7975dc/cartographer_ros/cartographer_ros/offline_node.cc#L172) for the offline node. It actually breaks a previously supported case, when you have one robot (i.e. same topics), and multiple sequential runs (bags).

[ERROR] [1517228479.547673264]: E0129 13:21:19.000000 29764 offline_node.cc:274] Sensor bag_2_scan resolves to topic /pioneer/scan which is already used by  sensor bag_1_scan

ojura, 2018-01-29:
I propose that we just change `bag_topic_to_sensor_id` from
`std::map<std::string,           cartographer::mapping::TrajectoryBuilderInterface::SensorId>`
to
`std::map<std::pair<size_t /* bag_index */, std::string>,
           cartographer::mapping::TrajectoryBuilderInterface::SensorId>`

gaschler, 2018-01-29:
(I'm not very familiar with this use case.)
What is the expected behavior?
Play all bags given by `bag_filenames` in a sequence?
Can you work around to achieve this?

Note that the error message alone is stricter than it needs to be, if that helps.

ojura, 2018-01-29:
> Play all bags given by bag_filenames in a sequence?

Yes, this was actually the only previous supported multi-trajectory behaviour of the offline node.  You could use only one configuration and set of topics with multiple sequential bags.

The bag multiplexer will correctly play the bags in a sequence; it's just that `bag_topic_to_sensor_id` messes up when multiple bags have the same topic. Adding the bag index to the map's key is an easy fix, I'll have the PR ready soon.

gaschler, 2018-01-29:
Sounds good, I can review it.
You could add more explanation to `bag_filenames` how one can use it.

PR #694 (2018-01-29): Offline node: better support for sequential bags.
Allow same topics to be used in different bags (a previously supported use case).
Remove unused variable `current_bag_sensor_topics`.
Touch up flag descriptions.

Fixes #693.

pair=@gaschler
commit 2538ac3e (2018-01-29): Offline node: better support for sequential bags. (#694)

Allow same topics to be used in different bags (a previously supported use case).
Remove unused variable `current_bag_sensor_topics`.
Touch up flag descriptions.

Fixes #693.

pair=@gaschler
closed 2018-01-29 by wally-the-cartographer