cartographer-project/cartographer_ros #695: How to use lifelong mapping feature?

Oluwatoni, 2018-01-29:
I looked through the google group messages and it wasn't clear to me if lifelong mapping has been implemented. If it has, how do I enable it for a robot running ROS?
@
Thank you!

guilhermelawless, 2018-01-30:
See at the bottom of https://github.com/googlecartographer/cartographer/projects

gaschler, 2018-01-30:
So far, the only part of the lifelong mapping feature that has been implemented is trimming arbitrary submaps from the pose graph. To use this, you need to implement a class derived from `mapping::PoseGraphTrimmer`.
This is not exposed to Ros.

(Closing as answered.)

closed 2018-01-30 by gaschler

musi00, 2018-02-05:
Does thi mean that cartographer's map currently grows infinitely if there is no submap deletion in place