cartographer-project/cartographer_ros #701: How can I use the Odom msgs?

Vincent1923, 2018-02-01:
I tested cartographer ROS using the Hokuyo UTM-30LX LiDAR (2d LiDAR) which published sensor_msgs/LaserScan and nav_msgs/Odometry msgs.  I have tried with the configuration files which I have modified. But when I wanted to use the nav_msgs/Odometry msgs in the cartographer, there were no map in the rviz. And this is the output of the terminal.

`[ WARN] [1517473382.001739492, 1517215329.645331520]: W0201 16:23:02.000000 74664 tf_bridge.cc:52] "base_footprint" passed to lookupTransform argument source_frame does not exist.`

Is there something wrong in my configuration files?

[The configuration file , launch files and urdf file is here.](https://gist.github.com/Vincent1923/02bda6247bcf7cff6f0cbea9d462394a)
And [this is the bag file I used](https://pan.baidu.com/s/1o9VWd1S).

Could you help me? Thank you!

BWT, the output of the bag file validate tool is:

`I0201 16:55:47.812906 75090 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser_link"):
Count: 3286  Min: 0.070574  Max: 0.079636  Mean: 0.075108
[0.070574, 0.071480)	                    	Count: 4 (0.121729%)	Total: 4 (0.121729%)
[0.071480, 0.072386)	                    	Count: 10 (0.304321%)	Total: 14 (0.426050%)
[0.072386, 0.073292)	                    	Count: 33 (1.004261%)	Total: 47 (1.430310%)
[0.073292, 0.074199)	                   #	Count: 131 (3.986610%)	Total: 178 (5.416920%)
[0.074199, 0.075105)	           #########	Count: 1404 (42.726719%)	Total: 1582 (48.143639%)
[0.075105, 0.076011)	           #########	Count: 1516 (46.135120%)	Total: 3098 (94.278755%)
[0.076011, 0.076918)	                   #	Count: 139 (4.230067%)	Total: 3237 (98.508827%)
[0.076918, 0.077824)	                    	Count: 34 (1.034693%)	Total: 3271 (99.543518%)
[0.077824, 0.078730)	                    	Count: 11 (0.334754%)	Total: 3282 (99.878273%)
[0.078730, 0.079636]	                    	Count: 4 (0.121729%)	Total: 3286 (100.000000%)
I0201 16:55:47.813580 75090 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
Count: 12345  Min: 0.000011  Max: 0.107410  Mean: 0.020000
[0.000011, 0.010751)	    ################	Count: 9876 (80.000000%)	Total: 9876 (80.000000%)
[0.010751, 0.021491)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.021491, 0.032230)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.032230, 0.042970)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.042970, 0.053710)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.053710, 0.064450)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.064450, 0.075190)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.075190, 0.085930)	                    	Count: 0 (0.000000%)	Total: 9876 (80.000000%)
[0.085930, 0.096670)	                   #	Count: 324 (2.624544%)	Total: 10200 (82.624542%)
[0.096670, 0.107410]	                 ###	Count: 2145 (17.375456%)	Total: 12345 (100.000000%)`

gaschler, 2018-02-01:
I'd assume that the odometry message child_frame_id does not exist in your urdf config. Inspect your bag what the frame names on the odom topic are and correct the urdf config.

What kind of sensor is that? If its odometry output is only based on 2d laser scans, it does not add any information.

BTW, rosbag_validate output looks fine.

Vincent1923, 2018-02-01:
I have changed the urdf config. But the result is not very well. I have no idea how to change the urdf to get a good map.

The odometry output is based on the wheels of the motor. Can I use the odometry output in the cartographer?

gaschler, 2018-02-01:
> But the result is not very well.

Sorry, transforms are still wrong. The only way to go forward is to disable global SLAM and correct the urdf.

> The odometry output is based on the wheels of the motor. Can I use the odometry output in the cartographer?

Sounds fine. Yeah, you can use it. Only if it is noisy, results may be better without.

Vincent1923, 2018-02-01:
Could you please tell me how to correct the urdf?

Vincent1923, 2018-02-07:
Hello! The child_frame_id of odom is "base_footprint". I have no idea how to change the URDF file to make cartographer use the odometry message. Could you please tell me how to correct the urdf config. @gaschler

Vincent1923, 2018-02-08:
Hello! I have changed the URDF file. But the result still didn't look good. 

If I didn't use the odom, the map was better than using the odom. 

I want to use the odometry message in the cartographer to get a better map. Could you give me some suggestions on what parameters I should be tuning or working with to make the map better? 

In addition, is there any mistake in my URDF configuration file?

Here is the map which I didn't use the odometry message to get.
[image]

And here is the map which I used the odometry message to get. The result is not good.
[image]

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler

Mafuqiang, 2019-05-29:
Do you fixed this problem? I have the same problem, and can you help me to fixed it?

Vincent1923, 2019-05-29:
I later found that my odom msg were inaccurate. So I'm not use odom msg anymore when I run cartographer ROS. Sorry about your question! This is another [issue ](https://github.com/googlecartographer/cartographer_ros/issues/792)from which you can get more information.

Mafuqiang, 2019-05-29:
I notice this issue, but I can not find the configuration file that you used like 'roslaunch cartographer_ros demo_hokuyo_2d_odom.launch bag_filename:=${HOME}/Downloads/2018-03-21-15-52-22.bag'. I want to know how to set the lua file. My cartographer result with odometry just like your first image showing. Thank you very much.

Mafuqiang, 2019-05-29:
I notice this issue, but I can not find the configuration file that you used like 'roslaunch cartographer_ros demo_hokuyo_2d_odom.launch bag_filename:=${HOME}/Downloads/2018-03-21-15-52-22.bag'. I want to know how to set the urdf file. My cartographer result with odometry just like your first image showing. Thank you very much.

Mafuqiang, 2019-05-29:
I can not open your link this issue 'The configuration file , launch files and urdf file is here.'.

Vincent1923, 2019-05-29:
I deleted the files before. But I've just uploaded the files again. You can take a look at it. Hope it will help you!