cartographer-project/cartographer_ros #706: Converting map images to Cartographer Sparse Pose Graph

musi00, 2018-02-05:
I'm just wondering if there is currently a way to convert .png map images to google cartographer sparse pose graphs.  If not, is this a feature planned in the future.  It would be useful to use cartographer's pure localization against maps generated by other SLAM algorithms which are usually stored as images.

ojura, 2018-02-05:
Hi! This is a feature that is not implemented yet. It's not trivial, because you would need to create a frozen trajectory (so the pose graph can be empty for that trajectory), with one large submap that would contain the map image (or imitating what Cartographer does, splitting the map image into a manyof overlapping submaps).

Most likely, this should be implemented as an "image-to-pbstream" tool. If you would like to move things forward in this direction, you should write a [request for comments](https://github.com/googlecartographer/rfcs).

gaschler, 2018-02-12:
This feature request sounds rather special to me.
Alternatively, you can discuss it in the open house to find out what the general interest is.

ojura, 2018-02-12:
@gaschler I've seen this feature or something similar (loading an png or pgm image as an initial map) requested a couple of times already. E.g. https://github.com/googlecartographer/cartographer/issues/255

At some (far) point in the future, I believe will be looking into something like this (e.g. loading a CAD to be used as a prior). I support if someone wants to jump into it sooner.

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler

vinaykumarhs2020, 2020-09-03:
@ojura Any developments on loading pgm/CAD floorplan? I was looking for something similar and would like to know if you have any suggestion on how to implement it.

MigVega, 2022-11-27:
Hi all,

Now you can use the [Ogm2Pgbm package](https://github.com/MigVega/Ogm2Pgbm/tree/main/ogm2pgbm) to do the conversion from .pgm to Pose graph based map in .pbtream format !

If haven't compiled cartographer already, you can follow the provided instructions in the [readme](https://github.com/MigVega/Ogm2Pgbm). It will allow you to compile everything in one isolated container.

Otherwise you can also use the package as a normal separated catkin package.

Have fun, and enjoy the power of graph-based localization! :D

P.S.: You can also convert from `.png` to `.pgm` using the open source image editor [GIMP](https://www.gimp.org/news/2022/06/14/gimp-2-10-32-released/)

MigVega, 2023-11-11:
The core of the package works independent of the ROS version.
However for the last step you need a trajectory builder, for that, we used either cartographer or Slam Toolbox.
So you have to consider the ROS version for which those packages are tested.
Nonetheless, the implementation in the repository allows you to build everything in a Docker container with ROS melodic (ROS 1) independent of the OS system you use.