cartographer-project/cartographer_ros #716: call /finish_trajectory results in 'std::out_of_range'

Joschka1, 2018-02-09:
If if run the demo data set with following command
`roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag`
and few moments later i call the rosservice 
`rosservice call /finish_trajectory "trajectory_id: 0"`
it results in an `std::out_of_range` error and the cartographer_offline_node die. 

```
rosservice call /finish_trajectory "trajectory_id: 0" 
status: 
  code: 0
  message: "Finished trajectory 0."

```
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
[cartographer_offline_node-3] process has died [pid 14081, exit code -6, cmd /home/joschka/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_offline_node -configuration_directory /home/joschka/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basenames backpack_2d.lua -urdf_filenames /home/joschka/catkin_ws/install_isolated/share/cartographer_ros/urdf/backpack_2d.urdf -bag_filenames /home/joschka/Downloads/b2-2016-04-05-14-44-52.bag echoes:=horizontal_laser_2d __name:=cartographer_offline_node __log:=/home/joschka/.ros/log/a49b6a3a-0d74-11e8-97f8-9061ae355c0a/cartographer_offline_node-3.log].
log file: /home/joschka/.ros/log/a49b6a3a-0d74-11e8-97f8-9061ae355c0a/cartographer_offline_node-3*.log
```

cschuet, 2018-02-09:
Finishing the trajectory during operation however works when you run the online node?

Joschka1, 2018-02-09:
I have not tried that yet.

LuffySSR, 2019-01-15:
@Joschka1 I got the same problem with you , did you remember how to deal with it?

maisa32, 2019-08-06:
@Joschka1 @cschuet Any update on this issue?

When running `roslaunch cartographer_ros offline_backpack_3d.launch bag_filenames:=${HOME}/Downloads/b3-2016-04-05-15-51-36.bag` and `rosservice call /finish_trajectory 0` I also get a
`terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at` and the cartographer process dies. `rosservice call /finish_trajectory 0
` reports `status: 
  code: 0
  message: "Finished trajectory 0."`

I am running Ubuntu 18.04 with ROS Melodic.

MichaelGrupp, 2020-10-21:
I haven't experienced this issue in recent versions. Feel free to reopen if it still happens.

closed 2020-10-21 by MichaelGrupp

qetuo105487900, 2024-10-07:
@Joschka1 I got the same problem with you