cartographer-project/cartographer_ros #718: question: pointers to information on workflow with install_isolated

faultdiagnosistoolbox, 2018-02-10:
Hi all, 

I tried out cartographer yesterday and it works great with data from my custom robot so my question is not about cartographer functionality directly.

My question instead concerns catkin_make_isolated which is new to me. I am no ROS wizard and have mainly used catkin-only packages and since cartographer requires catkin_make_isolated I no more understand the workflow and build process. 

After sourcing install_isolated/setup.basg and 'roscd my_package', that before led me to catkin_ws/src/my_package, I am now directed to catkin_ws/install_isolated/.../my_package
That directory is more or less empty and, for example, I can no longer run my ROS nodes as before.

Thus, it seems I don't really understand how to work with cartographer and my previous catkin-packages.  I have looked for information online but haven't been all that successful.

So, my question is: can anyone direct me to useful information about how I should work with install_isolated installed packages together with old-style catkin packages (I hope my question makes sense).

Thanks for open-sourcing this nice package.
  Erik

ojura, 2018-02-10:
Try sourcing devel_isolated/setup.bash.

Personally, I prefer the more modern `catkin build` over `catkin_make` and `catkin_make_isolated`. It's available in the package `python-catkin-tools`.

guilhermelawless, 2018-02-12:
@ojura is there anything similar to making isolated in python-catkin-tools? I'm also a big fan of it, but missing this feature.

ojura, 2018-02-12:
Actually, building isolated is the default mode of operation. I am not sure if it supports a non-isolated build at all.

faultdiagnosistoolbox, 2018-02-12:
Thanks for the reply.

I did source devel_isolated/setup.bash and I could then run cartographer, but then I could not access my other packages as I am used to.

But your help made me try out the new(?) build system ‘catkin build’ and now everything works as I expect it to and I am happy.

Thanks for the help,
Erik

[Linköping University]
Department of Electrical Engineering
SE-581 83 Linköping
Phone: +46 (0)13-28 57 14
Please visit us at www.liu.se<http://www.liu.se>

elgarbe, 2018-02-15:
Could you share your hardware/software setup? I feel that I'm missing some documentatio about how to config.

faultdiagnosistoolbox, 2018-02-18:
Hi, I'm not entirely sure what to tell you since it worked with very little problems for me. 

The hardware I use is a high-end toy RC-car that we equipped with an Arduino and a Raspberry Pi for controls, and magnets in the wheels to facilitate odometry. There is no IMU, only an RPLidar. The Raspberry runs ROS and all code serving the car (not that much) is developed from scratch by me. 

Cartographer kind of worked out of the box for me. The only thing I did not have at the first try was the correct transformations published (use tf2 library;  now I have three transformations that I publish:  map -> odom -> base_link -> laser_link), I didn't have the map->odom at first. Other than that, I publish odometry messages in a standard way and that did the trick for me.

Sorry, I cannot be more helpful.

cschuet, 2018-03-10:
Closing as answered.

closed 2018-03-10 by cschuet