cartographer-project/cartographer_ros #720: Getting 3D data

nitramek, 2018-02-13:
Hello,

I would like to make obstacle checker (via raycasting or something like that probably) for cartographer in 3D. I know you dont publicize OccupancyGridMap in 3D (via #330 ) I have tried to call submaps by their ids from SubmapList,  but it seems like there are only low and high resolution slices. 

My question is where I can get the 3D data from cartographer? Either in ROS way or within cartographer itself. 

Is it even possible to do now?
Thanks.

gaschler, 2018-02-14:
Currently, cartographer_ros does not publish 3D maps and there is no high-level interface.

Cartographer itself gives you access to all submap data with `PoseGraphInterface::GetAllSubmapData`.
Return data includes the protobuf data `cartographer.mapping.proto.Submap3D` which has `mapping_3d.proto.HybridGrid high_resolution_hybrid_grid`.

For offline processing, there is a nicer solution. You can use the asset writer to write a point cloud from the entire map.

(Closing as answered.)

closed 2018-02-14 by gaschler

andrecmilano, 2018-06-11:
Hello...

There is perhaps an update on this topic: "Publishment of a 3D global map - online". 

I am diving into the code to collect a global map and publish it continuously at a very low frequency and update it after pose optimization. I was curious if anyone else is already working on this issue. 

Thank you!

Sanderi44, 2018-06-19:
Hi Andre, have you have made any progress on this?  I would love to see what you have done!  Getting a 3D map in real-time would be perfect for my application.

Thanks!

andrecmilano, 2018-06-25:
Hello Sanderi,

I am working part-time on this project indeed the progress is quite slow.
Anyway, I am still struggling with this issue I am evaluating how useful it may be. At the end may not be necessary to publish the 3D map online, but I still have to figure it out how it is saved; if it is contained in the PoseGraph class or in MapBuilder. 

It may require few more weeks of tests and debug. 

Feel free to help me with this ;) I guess you are also working on SLAM with MAVs :)

Cheers

nitramek, 2018-06-25:
If this is any use to you I have created my fork or cartographer_ros and made similar service to `SubmapQuery` I call it `SubmapCloudQuery` and it fetches 3D data from Cartographer itself. I think it is not very efficient but maybe it will be useful to you guys somehow

https://github.com/nitramek/cartographer_ros/tree/submap_cloud_query

I publish one submap as `PointCloud2` which can be used for visualization in RViz in boxes or it can be processed with PCL anyway you want and because it is a service you can update that whenever you want.

Service signature: [SubmapCloudQuery.srv](https://github.com/nitramek/cartographer_ros/blob/submap_cloud_query/cartographer_ros_msgs/srv/SubmapCloudQuery.srv)
[Relavant code](https://github.com/nitramek/cartographer_ros/blob/submap_cloud_query/cartographer_ros/cartographer_ros/map_builder_bridge.cc#L192)

Also I publish the submaps it in fixed frame it was more useful for my use case. You need to transform it to global map location yourself.

andrecmilano, 2018-06-25:
Thank nitramek, I will check it! ;)

andrecmilano, 2018-07-02:
Hello @nitramek 

Finally I had time to test your code. When I run your service I receive this output:

> submap_version: 42
resolution: 0.0
finished: False
cloud: 
  header: 
    seq: 0
    stamp: 
      secs: 1528732114
      nsecs: 852897343
    frame_id: "map"
  height: 1
  width: 0
  fields: 
    - 
      name: "x"
      offset: 0
      datatype: 7
      count: 1
    - 
      name: "y"
      offset: 4
      datatype: 7
      count: 1
    - 
      name: "z"
      offset: 8
      datatype: 7
      count: 1
    - 
      name: "intensity"
      offset: 12
      datatype: 7
      count: 1
  is_bigendian: False
  point_step: 16
  row_step: 0
  data: []
  is_dense: Trueo.

I structured the request using these parameters:

> rosservice call /submap_cloud_query 
"trajectory_id: 0
submap_index: 1
min_probability: 0.0
high_resolution: true" 

Before to call your service I also called _/finished_trajectory_.

Unfortunatly I noticed that the "hybrid_grid.values_size()" returns zero, because cloud.width is zero.

Do I make any mistake? Do I miss something?

andrecmilano, 2018-07-02:
Update: Sorry... now I can read the data. I debugged a little and now it works, changed small things.

ojura, 2018-07-11:
@nitramek Nice work. It would be great if something like this was merged into the Submaps plugin.

nitramek, 2018-07-11:
Thanks @ojura, although I am not sure if this approach is good enough to be considered for merging. Also at the time I made it making it polished enough for PR was a last thing to think about.

ojura, 2018-07-12:
Even if you think it's not currently the best approach, it's still better than nothing :) I actually saw 3D submaps for the first time yesterday thanks to your hack. It really gives you a better insight into what's going on inside Cartographer.

Duke-Allen, 2019-03-03:
@nitramek 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?