cartographer-project/cartographer_ros #722: Minimum sensor and tf setup

elgarbe, 2018-02-15:
Hi, I have an ASV with Pixhawk and RPi 3.
Pixhawk provide a lot of msg by mavlink (mavros). RPi has a hokuyo lidar publishing laser scan data.
I just want to drive my boat, close to the coast, and record a bag file on Pi. Then download the bag file to my PC and run cartographer on my laptop to create the coast map. I don't want localization for now.
Is there a config procedure for that? What are the minimu sensor's, tf's, topics that cartogrpher need?
My laser is fixed to the ASV structure, the same for pixhawk. 
I'm using hector slam right now, but want to try cartographer as well.

Thank!

gaschler, 2018-02-16:
2D mapping needs a minimum of one laser scanner.
For low-end setups, double-check that the sensor timestamps (inside message headers) are correctly measured and recorded (and not just written from Time::now from a subscriber) and avoid frame drop (with buffering).

Cartographer works best if you see features around the entire horizon.
So I cannot guess how the results will be with seeing the coast only at some angle.

For a boat, IMU may help to correct the rotation and of course fixed frames from satellite navigation.
These are optional.

(Closing as answered, feel free to re-open.)

closed 2018-02-16 by gaschler

elgarbe, 2018-02-16:
So, I need a bag file with good laser scan data (I have an hokuyo lidar with urg_node) and optional IMU data (I have /mavros/imu/data and /mavros/imu/data_raw from pixhawk). I also have /mavros/odometry/odom from pixhawk.
Recording a bag file with that topics is all I need to run cartographer in my PC?
I would like to go to the river and record a bag file and later run cartogrpher on my PC.

Thank