cartographer-project/cartographer_ros #723: Performance estimation for multi-lidar at vertically low positions.

ozgudum, 2018-02-16:
I want to get your envision about what would be the mapping and localization performance of cartographer if we have 4 32 layer lidars mounted on the bumper and side mirror levels. Do you think we can map and localize in decent traffic scenario? For example vehicle is moving in 2 lane street and we have vehicle in front and back of the ego car. Second scenario would be to have vehicles in both sides of ego car. Any vehicle near the ego car will block the view of lidar on that side.

gaschler, 2018-02-16:
4 lidars are definitely supported.
I don't see any issue with the low position if you can see features in all directions most of the time.
You may want to tune for localization, see https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency.
Note that Cartographer makes no difference between features from static and moving obstacles so moving obstacles may create artifacts in the map.

(Closing as answered.)

ozgudum, 2018-02-16:
Thanks for your answer. Only major issue is the locations of the sensors. When they are mounted too low as opposed to mounting on top of the vehicle, surrounding cars can easily block most of the FOV of the sensor. And this will introduce a great dynamic change in the perceived world. If cartographer optimization algorithm is robust to these changes and can compensate those effects using sensor fusion, then I would be more optimistic about the performance. Because I haven't done much experiment using Cartographer, I can't assess the performance in such situations. 

Let me paraphrase my question. Has anyone tested Cartographer in highly dynamic scenes?

cschuet, 2018-02-17:
@ozgudum I think the short answer to your question is: "No", there have not been principled studies into Cartographer's localization performance in changing environments.

Our experience suggests however that Cartographer is in a good position to handle that case well. Cartographer running in pure localization mode is essentially the multi trajectory mapping case where the background map to localize against is represented as a frozen (i.e. not being optimized itself) trajectory and the "live" trajectory that is build from the incoming sensor data is clipped to constant length (i.e. Cartographer forgets submaps at the same pace as it builds new one's and the trajectory has a (roughly) constant length - think [Snake](http://ahumbleopinion.com/the-classic-snake-game/)). 

If your car ventures into an environment that has changed so much that loop closures against the (historic) background map cannot be found anymore, Cartographer falls back to local SLAM and tracking trajectory and background map start drifting apart at the rate that your local SLAM acquires error, which for well tuned sensors (wheel odometry, IMU, LiDAR) can be quite slow. Once it exits the unknown area and drives back into known territory the car "snaps back into place" with regards to the background map.

For your particular situation of a self-driving car in traffic, this reminds me of the "escalator problem" we had when using Cartographer for a LiDAR backpack for indoor mapping. Riding escalators posed the problem of nearby range-finder observations suggesting a state of rest (because the escalator stairs were apparently not moving) in contradiction to the integrated IMU measurements and maybe far-away features seen by the LiDAR. In early Cartographer versions however local features suggesting rest would win due to the large number of points they make up in the point clouds. This is why we introduced low- (for far-away features) and high-resolution (for nearby fearures) hybrid grids. 

If you are in a similar situation where the dynamicness of your scene is mainly due to nearby environmental changes, while far-away structures are rather stable (in time), you might want to try playing with the tuning of hi / low resolution grid weights for the scan matcher.

closed 2018-03-05 by cschuet