cartographer-project/cartographer_ros #727: F0217 20:38:43.000000 14475 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first

sauravag, 2018-02-18:
Howdy!

1. rosbag_validate output: https://gist.github.com/sauravag/cde85104206ecb6f542df13917e5ae74

2. Used the exact same configuration as cartographer_turtlebot

I'm running cartogrpaher_ros with a turtlebot in Gazebo. I have odometry and imu turned on. Cartographer node fails immediately on launch. I saw the same issue on the mailing list [here](https://groups.google.com/forum/#!topic/google-cartographer/b-NiyoRky2c).

I tried reducing the sampling ratio for odometry and imu but it still does not fix the issue for me. Any suggestions / fixes would be much appreciated!

dirtshell, 2018-02-20:
Do you have a rosbag to share? The devs haven't given a response, but I have a hunch this has to do with IMU messages with conflicting timestamps (see my issue with the same error: https://github.com/googlecartographer/cartographer_ros/issues/699). 

You can check pretty easily if you have overlapping timestamps. Simply call `rostopic echo /mobile_base/sensors/imu_data` when you are running your IMU drivers. Then check the timestamps in the headers.

cschuet, 2018-02-25:
Howdy! Can you share the complete stack trace and a sample rosbag file that causes this issue? That'd help us in debugging this.

closed 2018-03-10 by cschuet