cartographer-project/cartographer_ros #728: Cartographer2D using an IMU

KevinRiehl, 2018-02-19:
Dear Cartographer Team,
I achieved to get Cartographer running in 2D with VLP 16 Hires.
The results are awesome without IMU. Now I want to use my IMU 
(kobuki) but I have the same problem as before with my VLP 16 
Hires laserscan with tfs.

I try to run a modified version of "demo_backpack_2d.launch".
What should the IMUs topic name be? "/imu" ?
In which tf should it be published?

Can you please answer me these two questions?
Thank you very much,
KevinRiehl

gaschler, 2018-02-19:
> What should the IMUs topic name be? "/imu" ?

Right, cartographer_ros expects incoming IMU messages on topic "imu". You can remap in the launch file if your topic name is different.

> In which frame should it be published?

This depends on how you set up your URDF configuration.
In the examples, we use frame_id "imu_link".
I think you can directly use "base_link" if the IMU frame matches the robot frame.
If your IMU device is rotated with respect to your robot coordinates,
you can find an example setup in backpack_3d.urdf.

KevinRiehl, 2018-02-20:
Well I don't know what to do, to be honest.

Can you explain me, what the meaning of
"tracking_frame" and "published_frame" is?

In the situation of not using an IMU I had to set
both to "horizontal_laser_link", but normally it is
set to "base_link".

My kobuki already publishes on the topic and frame "/base_link",
this is why I have to change the cartographer code, because I cant
change the kobuki code. In fact I have a turtlebot with a VLP 16 Hires
put on its top.

Running my turtlebot bringup launchfile, I have following topics and frames:
IMPORTANT: It is just my kobuki / turtlebot, not the velodyne!

**rostopic list**
```/cmd_vel_mux/active
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/switch
/cmd_vel_mux/input/teleop
/cmd_vel_mux/parameter_descriptions
/cmd_vel_mux/parameter_updates
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/joint_states
/mobile_base/commands/controller_info
/mobile_base/commands/digital_output
/mobile_base/commands/external_power
/mobile_base/commands/led1
/mobile_base/commands/led2
/mobile_base/commands/motor_power
/mobile_base/commands/reset_odometry
/mobile_base/commands/sound
/mobile_base/commands/velocity
/mobile_base/controller_info
/mobile_base/debug/raw_control_command
/mobile_base/debug/raw_data_command
/mobile_base/debug/raw_data_stream
/mobile_base/events/bumper
/mobile_base/events/button
/mobile_base/events/cliff
/mobile_base/events/digital_input
/mobile_base/events/power_system
/mobile_base/events/robot_state
/mobile_base/events/wheel_drop
/mobile_base/sensors/bumper_pointcloud
/mobile_base/sensors/core
/mobile_base/sensors/dock_ir
/mobile_base/sensors/imu_data
/mobile_base/sensors/imu_data_raw
/mobile_base/version_info
/mobile_base_nodelet_manager/bond
/odom
/rosout
/rosout_agg
/tf
/tf_static
```
**rosrun tf tf_monitor**
```
RESULTS: for all Frames

Frames:
Frame: arm_base_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: base_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: camera_depth_frame published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
[... 37 lines cut ...]
Frame: standoff_50mm_11_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_1_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_2_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_3_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_4_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_5_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_6_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_7_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_8_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: standoff_50mm_9_link published by unknown_publisher(static) Average Delay: 0 Max Delay: 0

All Broadcasters:
Node: unknown_publisher(static) 1e+08 Hz, Average Delay: 0 Max Delay: 0
```

Which files do I have to change in order to get all my stuff running?
At the moment, I am just testing to change some parameters ins some files
in a random process until I find a configuration that works. :(
I dont even understand (in situation of not using IMU) why I have to set both
frames to "horizontal_laser_link" instead of "base_link".

Thanks for your great support so far,
without your help I would not have gotten realized my project without IMU so far.
Kind and hopefull regards,
Kevin

jimo123, 2018-03-31:
@KevinRiehl  Hello,
I'm using cartographer2D to build map with lidar and imu.when I run the node,the frame of "base_link" shakes so seriously so that it can build a integrated map.Can you please answer my question?
There is my .lua file,the tf,topic graph and map.

[iim_map_imu.lua.txt](https://github.com/googlecartographer/cartographer_ros/files/1865488/iim_map_imu.lua.txt)

[image]

[image]

[image]

Thanks for your help!!

gaschler, 2018-04-03:
Please provide urdf, lua, launch config as in fork so we can see the diff, following the issue template https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/.github/ISSUE_TEMPLATE

Is your IMU moving with respect to your laser sensor? If not, just set up a static config in urdf and do not publish these transforms.

> In the situation of not using an IMU (#725) I had to set both to "horizontal_laser_link", but normally it is set to "base_link".

You can choose the frame names yourself. They need to be consistent across message `frame_id`, the urdf and lua configs.

Why to you publish odom to base_link (by /robot_driver)? This transform should only be publish by cartographer_ros as an output.

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler