cartographer-project/cartographer_ros #729: how to use rosservice call /start_trajectory ?

tigerk0430, 2018-02-20:
hello,

I want to use /start_trajectory service call while using cartographer and add a new trajectory.

but It gives me 

```
E0220 11:13:15.379221 10766 node.cc:573] Invalid trajectory options.       
[ERROR] [1519092795.379307064]: E0220 11:13:15.000000 10766 node.cc:573] Invalid trajectory options.     

```

when I type 

```
rosservice call /start_trajectory "options: {tracking_frame: 'base_link', 
published_frame: 'base_link', odom_frame: 'odom', provide_odom_frame: false,  
use_odometry: false, use_nav_sat: false, num_laser_scans: 0, num_multi_echo_laser_scans: 0,  
num_subdivisions_per_laser_scan: 1, num_point_clouds: 1, rangefinder_sampling_ratio: 1.0,  
odometry_sampling_ratio: 1.0, fixed_frame_pose_sampling_ratio: 1.0,
 imu_sampling_ratio: 1.0,  
trajectory_builder_options_proto: ''}
topics: {laser_scan_topic: '/scan', multi_echo_laser_scan_topic: '/echoes', 
point_cloud2_topic: '/points2',  
imu_topic: '/imu', odometry_topic: '/odom', 
nav_sat_fix_topic: '/fix'}"
```

I guess It's because of null value of `trajectory_builder_options_proto: ''` 
how can I add this value? thanks

cschuet, 2018-02-22:
You have to specify the `trajectory_builder_options_proto` to start the trajectory with since the trajectory builder options can be different from trajectory to trajectory. The options you specify via the `.lua` files when starting the node are only used for the trajectories started with `StartTrajectoryWithDefaultTopics()`.

closed 2018-02-22 by cschuet

tigerk0430, 2018-02-23:
@cschuet  
Thanks to the reply,
that means I can't use /start_trajectory service call on terminal then I need to use it while using code.
that's clear Thx :-)