cartographer-project/cartographer_ros #731: 2D Cartographer with only one LIDAR and one IMU

jediofgever, 2018-02-22:
Hi There, 

I have searched around existing issues to find a answer for my question however I could not find some useful info, so please forgive me if this is already asked before. I have only one LIDAR that I would like to use with 2D Cartographer. I know that in Demos the bag files are recorded with 2 LIDARS, so when I record bag file with one LIDAR and one IMU it shows nothing on Rviz, 

WhaT part of code should be modified in order to be run the bag file I recorded with one LIDAR and IMU ?

Thank you !

ojura, 2018-02-22:
The second lidar (vertical laser) is not used at all during SLAM. Your problem lies somewhere else. Most basic problems are wrong frame names and topic remappings, make sure you got that right.

jediofgever, 2018-02-22:
Hi, @ojura 
Here is a screenshot  of my frames, Can you post your frames result so that I can compare where I got wrong?
[image]

ojura, 2018-02-22:
What is your tracking frame in cartographer? Is it `base_link`? You need a transform from it to the imu and laser_mount_link frames. Your tf tree has to be connected.

jediofgever, 2018-02-22:
Okay I think I understand where problem lays now. I am closing the issue, however I will comment if cannot get it work :) . Thanks for your help !

closed 2018-02-22 by jediofgever