cartographer-project/cartographer_ros #733: Build failure after last commit

SteveMacenski, 2018-02-22:
```
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartogra
pher_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /u
sr/include/eigen3 -isystem /home/simbe/catkin_ws/install_isolated/include -isyste
m /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../.
./include/xmlrpcpp -I. -I/home/simbe/catkin_ws/src/cartographer_ros/cartographer_
ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/incl
ude/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/in
clude/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -is
ystem /usr/local/include -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_
KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic 
-Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-typ
e -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartograp
her_ros.dir/cartographer_ros/ros_map_writing_points_processor.cc.o -MF CMakeFiles
/cartographer_ros.dir/cartographer_ros/ros_map_writing_points_processor.cc.o.d -o
 CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map_writing_points_processo
r.cc.o -c /home/simbe/catkin_ws/src/cartographer_ros/cartographer_ros/cartographe
r_ros/ros_map_writing_points_processor.cc
In file included from /home/simbe/catkin_ws/src/cartographer_ros/cartographer_ros
/cartographer_ros/ros_map_writing_points_processor.cc:17:0:
/home/simbe/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_
map_writing_points_processor.h:23:74: fatal error: cartographer/mapping_2d/proto/
range_data_inserter_options.pb.h: No such file or directory
compilation terminated.
```

Appears to be a result of the last commit from looking at TravisCI also seeing this.

ojura, 2018-02-22:
If I see everything correctly, there isn't a follow-up PR in this repo for the mapping_2d/3d -> mapping rename.

SteveMacenski, 2018-02-22:
I'm not familiar with proto. Is there a work around set of releases of cartographer/ros?

MichaelGrupp, 2018-02-23:
As of today, the only way to compile the current master of  `cartographer_ros` is to manually checkout the last compatible commit before "Purge 'mapping_2d::'." in `cartographer`. (see [git log](https://github.com/googlecartographer/cartographer/commits/master)).

Like @ojura said: as soon as the renaming also took place in `cartographer_ros` you can use the latest and greatest version of `master`again.

Ping @pifon2a

TobiMiller, 2018-02-23:
same here. just changing to old cartographer before "purge 'mapping_2d::" dont work

MichaelGrupp, 2018-02-23:
I didn't check which particular commit you'll need, just gave a quick hint, that's why I wrote last **compatible** commit. You should be on the safe side with the revision of February 12, as that was the last time that `cartographer_ros` made it hrough the CI pipeline.

cschuet, 2018-02-23:
Apologies for the inconvenience. This must have been overlooked. Unfortunately @pifon2a  is on vacation today. Let me try and find some time to send a follow-up PR today.

cschuet, 2018-02-23:
Once https://github.com/googlecartographer/cartographer_ros/pull/734 is merged we should be good.

cschuet, 2018-02-23:
Merged; should be good now. Please sync your clients to tot. Sorry again for the inconvenience.

SteveMacenski, 2018-02-23:
Awesome, I'm testing now and will close once I validate locally.

Thanks for being very attentive. If you have any documentation on the structure of the code base and areas of need, I'd love to find a way to get involved.

closed 2018-02-24 by SteveMacenski

pifon2a, 2018-02-26:
Sorry for inconvenience.

@cschuet: is it possible to set up some nagging email, when cartographer
changes break cartographer_ros?

cschuet, 2018-02-26:
No problem. You could check [here](https://docs.travis-ci.com/user/notifications/)