cartographer-project/cartographer_ros #735: export point cloud file .ply

sunmk2006, 2018-02-24:
hello,i am using online node of cartographer,i have no bag file,How to save the ply format of the point cloud file？

cschuet, 2018-02-25:
The online node does write the Cartographer state on exit, therefore you should make a ROS service all to `write_state` to generate a `.pbstream` file. You can then use asset writer and more specifically the `PlyWritingPointsProcessor` to generate a `.ply` file.

Here is an example [lua](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/assets_writer_backpack_2d.lua) and [launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/assets_writer_backpack_2d.launch) file.

closed 2018-02-25 by cschuet

sunmk2006, 2018-02-26:
the launch file needs[ a bagfile ](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/assets_writer_backpack_2d.launch#L24) , but i do not have,I'm using the online laser and imu data.

cschuet, 2018-02-26:
Right, sorry, I did not catch that. In that case there is no way. The `pbstream` only contains heavily filtered versions of the point cloud and this is not sufficient to generate a `.ply` file.