cartographer-project/cartographer_ros #742: Why i cannot compile cartographer_ros by catkin_make not use catkin_make_isolated? get some gtest errors

gangyahaidao, 2018-02-28:
Hello, i  have some problem about install cartographer_ros
on ubuntu 16.04, ros kinetic
when i use catkin_make -j8 to compile it , i got some errors:

```
-- Build type: Release
CMake Error at /usr/src/gtest/cmake/internal_utils.cmake:130 (add_library):
  add_library cannot create target "gtest" because another target with the
  same name already exists.  The existing target is a shared library created
  in source directory "/usr/src/gtest".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /usr/src/gtest/cmake/internal_utils.cmake:153 (cxx_library_with_type)
  /usr/src/gtest/CMakeLists.txt:70 (cxx_library)

CMake Error at /usr/src/gtest/cmake/internal_utils.cmake:130 (add_library):
  add_library cannot create target "gtest_main" because another target with
  the same name already exists.  The existing target is a shared library
  created in source directory "/usr/src/gtest".  See documentation for policy
  CMP0002 for more details.
Call Stack (most recent call first):
  /usr/src/gtest/cmake/internal_utils.cmake:153 (cxx_library_with_type)
  /usr/src/gtest/CMakeLists.txt:71 (cxx_library)

CMake Error at /usr/src/gtest/CMakeLists.txt:72 (target_link_libraries):
  Attempt to add link library "gtest" to target "gtest_main" which is not
  built in this directory.

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- looking for PCL_COMMON
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   iostreams
--   regex
-- +++ processing catkin package: 'cartographer_rviz'
-- ==> add_subdirectory(cartographer_ros/cartographer_rviz)
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /usr/local with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   iostreams
--   regex
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using Qt5 based on the rviz_QT_VERSION: 5.5.1
-- Configuring incomplete, errors occurred!
See also "/home/rosrobot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rosrobot/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:290: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

when i use: 
catkin_make_isolated --install --use-ninja 
this works fine.
but my other ros packages is compiled by catkin_make, so how can i compile cartographer_ros by catkin_make too??

ojura, 2018-02-28:
catkin_make does not support non-catkin CMake projects such as cartographer (not cartographer_ros) and ceres-solver.

My tip is to use `catkin build` (`python-catkin-tools` on ubuntu), which is better than both catkin_make and catkin_make_isolated

gangyahaidao, 2018-03-01:
@ojura Got it, Thanks a lot!

closed 2018-03-01 by gangyahaidao