cartographer-project/cartographer_ros #747: Is it better to do Gaussian blur to submap?

xlicz, 2018-03-01:
In all the other papers(e.g [1][2]) I've read about the correlative matching method, people always do Gaussian blur to reference scan first. This is because we need to consider the measurement noise. 
I'm wondering if it's better to do Gaussian blur to cartographer's submap. Or since the submap is already a probabilistic occupancy map with measurement noise considered, no need to do Gaussian blur, which would slow down speed? 

[1] Olson, E. (2009). Real-Time Correlative Scan Matching. Proc.~of the IEEE Int.~Conf.~on Robotics & Automation (ICRA). 
[2] Olson, E. (2015). M3RSM: Many-to-many multi-resolution scan matching. In Proceedings - IEEE International Conference on Robotics and Automation (Vol. 2015–June).

gaschler, 2018-03-01:
I'm not sure if this will improve quality, it will certainly make submap insertion much slower.
Our submaps are probabilistic (16 bit resolution) occupancy grids in both 2D and 3D. In 2D, the full ray is inserted with subsampling (think of an anti-aliased line). In 3D, a short `num_free_space_voxels` length line is inserted together with the hit.

However, I remember that in voxel filtering, results are worse when you take the average or interpolation of multiple measurements. In the filtering step, it's better to only throw out some returns.

So, in case you want to run this experiment with blurring submaps, I'd be interested to see the results.

xlicz, 2018-03-02:
@damonkohler @SirVer How do you consider the blur or not blur problem?

wohe, 2018-03-08:
+1 to what @gaschler said. Without a demonstrated advantage in quality, it is only clear that it would reduce performance. Also note that the papers are concerned with scan-to-scan matching which can be different from scan-to-map like we do. There might be cases where the sensor noise can be larger than the submap resolution, e.g. depth cameras at longer distance. But even in this case it seems to me the performance implications have to be considered very carefully.

closed 2018-03-10 by cschuet