cartographer-project/cartographer_ros #759: Adding categorized but not unique landmarks

Magnusgaertner, 2018-03-07:
Is is currently possible to add landmarks without a unique identifier? Is it moreover possible to add Landmarks that belong to a category?

gaschler, 2018-03-08:
Landmarks are observations of the same object at different times, and are used for global optimization, so they need to be identified.
If you are thinking of adding annotations, this is currently not supported, and is not the purpose of `LandmarkData`.

Magnusgaertner, 2018-03-08:
Then not using the LandmarkData, how would one link a locally seen object to the cartographer map so that one could compute the optimized pose of the observed object after a map optimization done by the cartographer. Should the local observation be represented in the submap frame as the transformation of submaps with respect to the world gets optimized.
The computation would be:
map->submap->observed object.
Or should one link to the tracked frame with a timestamp to get the optimization by map-> trajectory(at timestamp) -> observed object.
One should mention that the observed objects are assumed to be static in the fixed frame.

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler

reinzor, 2019-11-07:
I would like to continue up-on this request. If landmark data require an unique identifier, we need data association if we cannot extract the unique identifier from the landmark measurement (e.g. measured reflector). What would be the way to go to integrate reflector measurements into Google Cartographer? Any suggestions?