cartographer-project/cartographer_ros #764: No [FATAL] message on console

javirrs, 2018-03-08:
We are trying to run cartographer with a simulation on Gazebo. As a laser scan we are using a Hokuyo.
When doing roslaunch, the cartographer_node crashes and we get the following:

```
[ INFO] [1520536704.546934485]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1520536704.547816687]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1520536704.547947247]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1520536704.548127790]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1520536704.548217069]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1520536704.549287719]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1520536704.549393510]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1520536704.549545925]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1520536704.549633349]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1520536704.549900899]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1520536704.549998082]: I0308 20:18:24.000000  4066 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1520536704.762529171]: I0308 20:18:24.000000  4066 submaps.cc:203] Added submap 1
[ INFO] [1520536704.762808978]: I0308 20:18:24.000000  4066 map_builder_bridge.cc:120] Added trajectory with ID '0'.
[ INFO] [1520536705.072029060]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1520536705.077492548]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
SpawnModel script started
[INFO] [1520536706.393808, 0.000000]: Loading model XML from ros parameter
[INFO] [1520536706.410103, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1520536706.742127, 0.010000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1520536706.764906590, 0.031000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1520536707.114132, 0.185000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1520536707.221079049, 0.185000000]: Physics dynamic reconfigure ready.
[ INFO] [1520536707.764103761, 0.185000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1520536707.764358832, 0.185000000]: Starting GazeboRosLaser Plugin (ns = /)
[ INFO] [1520536707.786382381, 0.185000000]: GPU Laser Plugin (ns = /) <tf_prefix_>, set to ""
[ INFO] [1520536707.794455581, 0.185000000]: LoadThread function completed
[ INFO] [1520536707.836203397, 0.185000000]: Starting plugin DiffDrive(ns = //)
[ WARN] [1520536707.836709762, 0.185000000]: DiffDrive(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1520536707.842449792, 0.185000000]: DiffDrive(ns = //): <tf_prefix> = 
[ WARN] [1520536707.843367387, 0.185000000]: DiffDrive(ns = //): missing <publishWheelTF> default is false
[ WARN] [1520536707.843636953, 0.185000000]: DiffDrive(ns = //): missing <publishWheelJointState> default is false
[ WARN] [1520536707.844442580, 0.185000000]: DiffDrive(ns = //): missing <wheelAcceleration> default is 0
[ WARN] [1520536707.844654323, 0.185000000]: DiffDrive(ns = //): missing <wheelTorque> default is 5
[ WARN] [1520536707.845557102, 0.185000000]: DiffDrive(ns = //): missing <odometrySource> default is 1
[ WARN] [1520536707.852407461, 0.185000000]: GazeboRosDiffDrive Plugin (ns = ) missing <publishTf>, defaults to 1
[ INFO] [1520536707.887288822, 0.185000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1520536707.921187117, 0.185000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1520536707.938316711, 0.185000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1520536707.980551020, 0.206000000]: imu plugin missing <xyzOffset>, defaults to 0s
[ INFO] [1520536707.983239612, 0.209000000]: imu plugin missing <rpyOffset>, defaults to 0s
[bot_spawn-6] process has finished cleanly
log file: /home/nvidia/.ros/log/77cefa5c-2305-11e8-a088-00044b8d78f3/bot_spawn-6*.log
[cartographer_node-4] process has died [pid 4066, exit code -11, cmd /home/nvidia/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/nvidia/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua __name:=cartographer_node __log:=/home/nvidia/.ros/log/77cefa5c-2305-11e8-a088-00044b8d78f3/cartographer_node-4.log].
log file: /home/nvidia/.ros/log/77cefa5c-2305-11e8-a088-00044b8d78f3/cartographer_node-4*.log
```

Since we get no fatal message we are unclear about what went wrong.

gaschler, 2018-03-09:
Exit code is -11, segfault.
Can you please re-run inside a debugger, ideally with debug symbols compiled in? 
If you think this helps, you can also share your launch and config files.

javirrs, 2018-03-09:
This is the error that we got from gdb:
```
0x00000000005b6cec in cartographer_ros::SensorBridge::HandleLaserScan(std::__cxx11::basic_string<char, 
std::char_traits<char>, std::allocator<char> > const&, 
std::chrono::time_point<cartographer::common::UniversalTimeScaleClock, std::chrono::duration<long, 
std::ratio<1l, 10000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, 
std::allocator<char> > const&, cartographer::sensor::PointCloudWithIntensities const&) ()
```
Our launch and config files:
[launch.txt](https://github.com/googlecartographer/cartographer_ros/files/1797780/launch.txt)
[backpack_2d.lua.txt](https://github.com/googlecartographer/cartographer_ros/files/1797788/backpack_2d.lua.txt)

gaschler, 2018-03-12:
Sorry, looking at `HandleLaserScan` and the files you provided it is not clear to me what the cause of this segfault is.
Maybe you can do some printf-debugging in `SensorBridge::HandleLaserScan`, printing out the size of points and so on?

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler