cartographer-project/cartographer_ros #765: CTEST_OUTPUT_ON_FAILURE=1 ninja test Failed

YanhaoZhang, 2018-03-10:
Hi. 
Thank you for providing this system. I am trying to install the not ros version, and I get stuck when running CTEST_OUTPUT_ON_FAILURE=1 ninja test Failed. The results are
The following tests FAILED:
	 19 - cartographer.mapping.2d.pose_graph_2d_test (OTHER_FAULT)
	 30 - cartographer.mapping.internal.3d.local_trajectory_builder_3d_test (OTHER_FAULT)
	 36 - cartographer.mapping.map_builder_test (OTHER_FAULT)
Errors while running CTest
ninja: build stopped: subcommand failed.
For more details:

1. [1/1] Running tests...*** Error in `/home/yanhao/Programs/cartographer-master/build/cartographer.mapping.2d.pose_graph_2d_test': free(): invalid pointer: 0x0000000000b530d8 ***
2. *** Error in `/home/yanhao/Programs/cartographer-master/build/cartographer.mapping.internal.3d.local_trajectory_builder_3d_test': free(): invalid pointer: 0x0000000000afe078 ***
3. *** Error in `/home/yanhao/Programs/cartographer-master/build/cartographer.mapping.map_builder_test': free(): invalid pointer: 0x0000000000bd1138 ***
4. [1/1] Running tests...
5. FAILED: cd /home/yanhao/Programs/cartographer-master/build && /usr/bin/ctest --force-new-ctest-process

Could you please give me some idea of this issue? 

Another thing is you forget to put file 'FindEigen3.cmake' in '/cmake/modules/'. 

I am a beginner of ros and programming. I would like to know it is better for me to start from your ros version of cartographer, is it? Thank you very much. 

Cheers,
Yanhao Zhang

cschuet, 2018-03-10:
The non-ROS version is just a library. If you want to e.g. run one of our example bags to generate a map you want to install cartographer_ros. Please  follow the build instructions as outlined in the readme. You will be using ROS's catkin build system.

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler