cartographer-project/cartographer_ros #766: Cartographer2D crashes

GMahmoud, 2018-03-12:
Dear Team,

I am using catographer2D on a Kobuki platform with a  Hokuyo URG-04LX-UG01 Laser . 
When the robot finishes its first loop, cartographer crashes.
Here's the last Info before crash:
```[ INFO] [1520853984.661969578, 1520850612.346367899]: I0312 12:26:24.000000 24626 constraint_builder_2d.cc:257] Node (0, 398) with 8 points on submap (0, 2) differs by translation 6.57 rotation 0.194 with score 90.0%.```
I suppose that I'm not using the correct lua params for my platform.  I saved a rosbag for this issue if you could help me to correct it.

BagFile:
https://www.dropbox.com/s/0exj8mex90yob8r/data-for-mapping-coworking_2018-03-12-11-29-08.bag?dl=0
LuaConfig:
[emx.lua.txt](https://github.com/googlecartographer/cartographer_ros/files/1802689/emx.lua.txt)
 
Thanks
MG

gaschler, 2018-03-12:
Please re-run cartographer_node in a debugger. (The info line is unrelated.)

GMahmoud, 2018-03-12:
Here's what gdb returns: 
```
Program received signal SIGSEGV, Segmentation fault.
0x00000000005b2f9f in cartographer_ros::SensorBridge::HandleLaserScan(std::string const&, std::chrono::time_point<cartographer::common::UniversalTimeScaleClock, std::chrono::duration<long, std::ratio<1l, 10000000l> > >, std::string const&, cartographer::sensor::PointCloudWithIntensities const&) ()
(gdb) bt
#0  0x00000000005b2f9f in cartographer_ros::SensorBridge::HandleLaserScan(std::string const&, std::chrono::time_point<cartographer::common::UniversalTimeScaleClock, std::chrono::duration<long, std::ratio<1l, 10000000l> > >, std::string const&, cartographer::sensor::PointCloudWithIntensities const&) ()
#1  0x00000000005b3397 in cartographer_ros::SensorBridge::HandleLaserScanMessage(std::string const&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&) ()
#2  0x000000000058b3da in cartographer_ros::Node::HandleLaserScanMessage(int, std::string const&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)
    ()
#3  0x000000000059912c in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const>) ()
#4  0x00000000005a8275 in ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
#5  0x00007ffff557a645 in ros::SubscriptionQueue::call() () from /opt/ros/indigo/lib/libroscpp.so
#6  0x00007ffff5534107 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/indigo/lib/libroscpp.so
#7  0x00007ffff5534c53 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/indigo/lib/libroscpp.so
#8  0x00007ffff557d175 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/indigo/lib/libroscpp.so
#9  0x00007ffff5564d9b in ros::spin() () from /opt/ros/indigo/lib/libroscpp.so
#10 0x000000000058866b in cartographer_ros::(anonymous namespace)::Run() ()
#11 0x0000000000585bd6 in main ()
```

gaschler, 2018-03-12:
Thanks for your quick response. This seems similar to https://github.com/googlecartographer/cartographer_ros/issues/764.
Can you please make this reproducible by creating a fork repository containing a branch of    `cartographer_ros` containing all the configuration, launch, and URDF files required?

GMahmoud, 2018-03-12:
You can find in the rosbag file all topics you need. I deleted the /tf (map -> odom) coming from cartographer_node when I saved the bag. So I attached a new one (output.bag)
link: https://www.dropbox.com/s/oy7wkjkv6jiy8ft/output.bag?dl=0
To run my bag just launch the ```play.launch```. You have to put your directory path in this launch file.

[play.launch.txt](https://github.com/googlecartographer/cartographer_ros/files/1802995/play.launch.txt)
[emx.launch.txt](https://github.com/googlecartographer/cartographer_ros/files/1802996/emx.launch.txt)
[emx.lua.txt](https://github.com/googlecartographer/cartographer_ros/files/1802997/emx.lua.txt)

I did some experimentation with my robot and I think that the crash happens when the laser is not detecting any obstacle (big space). I am checking that case for now.
  
Cheers,
MG

ojura, 2018-03-12:
Could it be that you're passing a completely empty rangefinder measurement (all rays are out of range, i.e. the resulting point cloud is empty)?

gaschler, 2018-03-12:
@GMahmoud could you move the check `CHECK_LE(points.points.back()[3], 0);` in `sensor_bridge.cc` after the lines `if (start_index == end_index) { continue; }` and see if this works?

GMahmoud, 2018-03-12:
Yeah effectively. When the vector of points of ```PointCloudWithIntensities``` is empty, the ```CHECK_LE(points.points.back()[3], 0);``` is sending a Segmentation fault.

GMahmoud, 2018-03-12:
The problem is solved with your suggestion @gaschler.
Thanks

cschuet, 2018-04-19:
Closing as solved.

closed 2018-04-19 by cschuet