cartographer-project/cartographer_ros #769: Regression: build jobs fail due to refactoring in cartographer

MichaelGrupp, 2018-03-13:
Hi guys,

you're probably already aware of this, but the CI jobs of `cartographer_ros` and all dependent `cartographer_*` repos fail due to the recent API changes without follow-ups in `cartographer_ros`. Consequently the CQ pipeline too. @pifon2a 

Not super urgent from my side, I just got a notification from our daily build of `cartographer_magazino`.

pifon2a, 2018-03-13:
I am on it. Thanks.

pifon2a, 2018-03-13:
Surprisingly enough, this is caused not by the API changes. One problem is gmock/gtest problems with catkin. @gaschler has already provided a fix. (https://github.com/googlecartographer/cartographer_ros/pull/768).

MichaelGrupp, 2018-03-13:
Ok, I changed the title ;) You can close this then.

closed 2018-03-13 by pifon2a