cartographer-project/cartographer_ros #774: [] Create standalone package with snapcraft and Cartographer

DaniloCaporale, 2018-03-16:
Hi everyone,
first of all, thanks for the great work you're doing with Cartographer. In my university lab we love this software and are considering using it for a variety of projects.

I'm writing for something not quite related to the software development but more related to its deployment on robots.

We are testing snapcraft to deploy code on several robots in a simple way, for distributed robotic platforms this would be really an interesting topic.

Right now I'm stuck as I can't build a simple snap with just cartographer inside.
I created a snapcraft.yaml file which contains the following:

```
name: test # you probably want to 'snapcraft register <name>'
version: '0.1' # just for humans, typically '1.2+git' or '1.3.2'
summary: test snap build for cartographer
description: |
  Foobar

grade: devel # must be 'stable' to release into candidate/stable channels
confinement: devmode # use 'strict' once you have the right plugs and slots

protobuf:
    plugin: cmake
    source: "https://github.com/google/protobuf.git"
    source-tag: "v3.4.1"
    build: |
        cmake -G Ninja -DCMAKE_INSTALL_PREFIX=$SNAPCRAFT_STAGE/ -DCMAKE_POSITION_INDEPENDENT_CODE=ON -DCMAKE_BUILD_TYPE=Release -Dprotobuf_BUILD_TESTS=OFF ../src/cmake
        ninja
    install: |
        ninja install
    
workspace:
    plugin: catkin
    rosdistro: kinetic
    catkin-packages: [ceres-solver, cartographer_ros_msgs, cartographer_ros, cartographer]
    after: [protobuf]
    build: |
        rosdep update
        rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
        catkin_make_isolated --install --use-ninja
```

I followed the instructions from your read the docs, building with `catkin_make_isolated...` is perfectly fine. When I run `snapcraft` on this file, building always fails when building the last package (cartographer), throwing errors about protobuf versions.

Did you ever try anything like this? Would you have any suggestions?
I would appreciate any kind of help. On a side note, if you never tried snapcraft, I'll be glad to provide feedback on its performance and effectiveness :)

Cheers,

Danilo

gaschler, 2018-03-19:
Sounds like cmake finds only the OS protobuf, which is too old. Make sure that `-DCMAKE_INSTALL_PREFIX=$SNAPCRAFT_STAGE/` does the right thing.
(Cartographer requires Protobuf 3 .)

DaniloCaporale, 2018-03-19:
Thanks @gaschler . Yes it should be that. I noticed errors due to protobuf 2 instead of 3 being used to build cartographer.

Protobuf looks installed in `$SNAPCRAFT_STAGE/` as in `stage` folder I can see:
```
$ ls *

bin:
protoc

include:
google

lib:
cmake  libprotobuf.a  libprotobuf-lite.a  libprotoc.a  pkgconfig
```

Still, it looks like cartographer uses the older OS version.
Cmake though gives this:
```
-- Found Protobuf: /home/danilo/catkin_ws/stage/lib/libprotobuf.a (Required is at least version "3.0.0") 
```
and I edited CMakeLists.txt to show the following messages:
```
#################################################################
PROTOBUF_INCLUDE_DIR
/home/danilo/catkin_ws/stage/include
PROTOBUF_PROTOC_LIBRARY
/home/danilo/catkin_ws/stage/lib/libprotoc.a
#################################################################
```
which looks correct to me.

gaschler, 2018-05-02:
Closing for inactivity, feel free to post when there is something new.

closed 2018-05-02 by gaschler

zchao9456, 2018-12-25:
> Thanks @gaschler . Yes it should be that. I noticed errors due to protobuf 2 instead of 3 being used to build cartographer.
>
> Protobuf looks installed in `$SNAPCRAFT_STAGE/` as in `stage` folder I [...]

@DaniloCaporale  hello,Have you solved the problem yet?My problem is the same as yours.can you help me,thank you very much