cartographer-project/cartographer_ros #776: Cannot configure gmock in Lunar

gaschler, 2018-03-19:
How to reproduce: `docker build -f Dockerfile.lunar .`

Error:
```
-- Found gmock: gmock and gtests will be built
CMake Error at /opt/ros/lunar/share/catkin/cmake/test/gtest.cmake:381 (add_library):
  add_library cannot create imported target "gmock" because another target
  with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/lunar/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/lunar/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:45 (find_package)
CMake Error at /opt/ros/lunar/share/catkin/cmake/test/gtest.cmake:383 (add_library):
  add_library cannot create imported target "gmock_main" because another
  target with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/lunar/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/lunar/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:45 (find_package)
```

I tried several ways to fix cartographer_ros' CMakeLists.txt alone, and also in the FindGMock.cmake of cartographer. However,

- Moving the catkin find_package earlier as in https://github.com/googlecartographer/cartographer_ros/pull/768 will raise "add_library cannot create target "gtest" because another target with the same name already exists" in gmock itself (when it is built by our FindGMock.cmake).
- Disabling our `find_package(GMock REQUIRED)` will later lead to linker issues. (Perhaps we require static linking?)

Up to ROS Kinetic, we had the workflow that cartographer_ros calls https://github.com/googlecartographer/cartographer/blob/master/cmake/modules/FindGMock.cmake#L58
which usually compiles `usr/src/gmock` (in add_subdirectory, which creates a target `gmock_main` in our scope that we link).
Now, newer versions of ROS Lunar create an imported target "gmock(_main)", which leads to a name conflict.
https://github.com/ros/catkin/blob/kinetic-devel/cmake/test/gtest.cmake#L381

@dirk-thomas Thanks for your comment on my PR. Do you have an idea how to resolve this? (I'm out of ideas in this case.)

ojura, 2018-03-19:
@gaschler I have no such problems building on Lunar with my setup, and that is having a CMake project with both cartographer and cartographer_ros. In there, I find_package(GMock) before catkin, and it works.

https://github.com/larics/cartographer_superbuild/blob/master/CMakeLists.txt

ojura, 2018-03-20:
Actually, I just found out that my approach fails with kinetic:
```
-- Found gmock: gmock and gtests will be built
CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:381 (add_library):
  add_library cannot create imported target "gmock" because another target
  with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  cartographer_ros/cartographer_rviz/CMakeLists.txt:35 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:383 (add_library):
  add_library cannot create imported target "gmock_main" because another
  target with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  cartographer_ros/cartographer_rviz/CMakeLists.txt:35 (find_package)
```

ojura, 2018-03-20:
Okay, I think I see what is going on. When you start with a clean build folder, Cartographer's `FindGMock.cmake` calls `find_library(GMOCK_LIBRARIES ..)`, and the result is `GMOCK_LIBRARIES=GMOCK_LIBRARIES-NOTFOUND`. This triggers entering the if branch (`if(NOT GMOCK_LIBRARIES)`), where the gmock subdirectory is added, and it defines its `gmock` and `gmock_main` targets. Afterwards, Catkin _does not_ detect GMock added this way, and tries adding its own `gmock/gmock_main`, which fails.

However, if you invoke CMake _again_ in the same build directory, the find_library call in `FindGMock.cmake` will set  `GMOCK_LIBRARIES` to `gmock_main;-lpthread;-lpthread`, the `if` branch will not be entered, so Catkin will happily add its own `gmock` and `gmock_main`.  Notice that Cartographer's `FindGMock.cmake` spams `-lpthread` at the end of `GMOCK_LIBRARIES`, which does not seem as intended behaviour.

I think Cartographer's `FindGMock.cmake` needs to be updated to play nice with Catkin, and to avoid unintended behaviour (different logic in different runs, and adding an additional `-lpthread` in each run).

ojura, 2018-03-20:
I think it would make sense that Catkin does not `add_library()` `gmock` and `gmock_main` if `find_package(GMock)` (i.e. Cartographer's `FindGMock.cmake`) already caused these libraries to be added. https://github.com/ros/catkin/pull/927

ojura, 2018-03-20:
Okay everyone, sorry for spamming. Here's the follow-up PR for Cartographer's FindGMock.cmake: https://github.com/googlecartographer/cartographer/pull/1011

JohannesBetz, 2018-03-25:
Hey everyone,

i have the same issue with ROS Kinetic on my AMD64 System. I doublechecked with an ARM system and got no issue there. But when i try to build isolated on my AMD system, i get the same error as shown above.
Do we have to wait that everything is merged into the branch? What do you suggest ?

ojura, 2018-03-25:
@JohannesBetz there are a couple of options:
- apply the patch above to Catkin:
https://github.com/ros/catkin/commit/38b85c6f814dfc46843bb22e78e4ca995b1db4ce
- apply https://github.com/googlecartographer/cartographer/pull/1011:
- build cartographer_ros without tests (call `catkin_make_isolated` with  `--cmake-args -DCATKIN_ENABLE_TESTING=false`)

gaschler, 2018-04-03:
Fixed by https://github.com/googlecartographer/cartographer/pull/1011

closed 2018-04-03 by gaschler