cartographer-project/cartographer_ros #778: Ideal Sensor Suite for 3D Mapping and Localization

saimanoj18, 2018-03-23:
I am re-posting the question from as this group may have people with more experience... https://github.com/googlecartographer/cartographer/issues/1014

Thanks for this amazing package.

We are planning to employ Google Cartographer for 3D mapping of outdoor large scale environments of few kilometers for self driving car localization and related applications.

We are yet to buy the sensors and from your experience, which sensors and their configuration works best for cartographer ??

    Which IMUs should we prefer? Could you give us some specifications such as bias or error ?
    It was said that two velodyne 16's are a good choice and work out of the box. Can someone elaborate on this ? Is one Velodyne 32 is better than two Velo 16's ?
    What kind of configuration should be have ?(One at the front of the vehicle and other at back with tilted angles ?)

3.It is said that we should constrain it in 6 DOF. Could you say, in which way do we miss the aspect of constraining if we use one Velo 16 and how does that get nullified when we use two Velo 16's.

Thank you so much!

saimanoj18, 2018-03-28:
Thank you @dirtshell

Entropy512, 2018-04-03:
As an FYI, on a vehicle that I'm hoping to start testing soon (I haven't had a chance to touch Cartographer for around 4+ months), we have the two VLP-16s mounted at 20 degree angles tilted downwards to the sides of the vehicle.  The idea being that the -15 beam from each unit should clear the other one, and now we can start seeing ceiling/floor in many situations.

As to IMU - I sort of recall hearing (indirectly) that some of their platforms use Microstrain IMUs.  These seem like a fairly good bet, however there seem to be 2-3+ different ROS nodes for talking to Microstrain IMUs and I'm not sure which one is ideal.

closed 2020-10-21 by MichaelGrupp