cartographer-project/cartographer_ros #779: Problem during the installation

Thomas1994Johansen, 2018-03-23:
Hi
I am trying to install the cartographer package for ros on my ubuntu 16.04 virtual machine. I have installed the kinetic ros package and followed the instructions provided in this tutorial https://google-cartographer-ros.readthedocs.io/en/latest/index.html to install cartographer with the exception of creating the workspace since i have all ready done that when installing ROS as described at the end of this post. 

When i type the following command 
catkin_make_isolated --install --use-ninja
in my terminal it produces this error message when processing the cartographer_ros package

https://drive.google.com/open?id=1UcXGtCoR6Ogi3zPKvYbOj6VxkyP2UFz0wu78TDRHvPE

I first tried on one machine were i first installed two other ros packages and got that error, so then i tried to install the cartographer right after completing the ros kinetic installation on a clean new virtual machine, but that produced the same error. 

When installing ros kinetic i follow this tutorial http://wiki.ros.org/kinetic/Installation/Ubuntu
and then to create my workspace i type 

sudo apt-get install python-catkin-tools python-rosinstall-generator -y

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src

ojura, 2018-03-23:
Either add `--cmake-args -DCATKIN_ENABLE_TESTING=false` to invocation of `catkin_make_isolated`, or apply this patch to your ROS installation:

https://github.com/ros/catkin/pull/927/files

Catkin currently breaks cartographer, we are waiting for a release which include the above patch.

Thomas1994Johansen, 2018-03-23:
That worked.

closed 2018-03-23 by Thomas1994Johansen