cartographer-project/cartographer_ros #782: Recording a bag file for catographer

elgarbe, 2018-03-24:
you are closing my issues before I get it working!
this is related to https://github.com/googlecartographer/cartographer_ros/issues/726

My hokuyo LIDAR an all of my system is working ok with hector slam.
To get it working I run urg_node on the raspberry pi (mounted on my boat) and this topics:

/diagnostics
/laser_status
/rosout
/rosout_agg
/scan
/urg_node/parameter_descriptions
/urg_node/parameter_updates

then, in another terminal I run rosbag record -a

I slowly drive the ASV as close to the coast as possible.

Then copy bag file to my PC and run hector slam.
So, that way there is not tf and there is not tf static. 

My first question is what do I need to add to my basic setup to get a valid bag file to work with cartographer with just LIDAR data in OFFLINE mode.

My LIDAR is mounted rotated 90º CW (pointing to +Y axis of the ASV). This way laser beam is pointing to one coast (lidar max distance is 4mts, so there is no way to get 2 coast on sight).

cschuet, 2018-03-24:
Can you share a bag file? Then we can take a look.

elgarbe, 2018-03-24:
[here](https://www.dropbox.com/s/pnxtey99e56mo7v/19_03_18_01.bag?dl=0) you can find the last experiment.

cschuet, 2018-03-26:
Thanks for sharing the data. I got it to SLAM by starting from the backpack_2d.lua and changing a couple of frame names, so bag is perfectly valid. I ran the bag through our cartographer_rosbag_validate which checks for most common errors and the output

```
E0326 09:03:01.140959 161081 rosbag_validate_main.cc:374] Point data (frame_id: "laser") has a large gap, largest is 0.102 s, recommended is [0.0005, 0.05] s with no jitter.
I0326 09:03:01.141556 161081 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser"):
Count: 1524  Min: 0.098205  Max: 0.102000  Mean: 0.099787
[0.098205, 0.098584)	                    	Count: 24 (1.574803%)	Total: 24 (1.574803%)
[0.098584, 0.098964)	                    	Count: 0 (0.000000%)	Total: 24 (1.574803%)
[0.098964, 0.099343)	                    	Count: 8 (0.524934%)	Total: 32 (2.099738%)
[0.099343, 0.099723)	     ###############	Count: 1115 (73.162727%)	Total: 1147 (75.262466%)
[0.099723, 0.100102)	                    	Count: 5 (0.328084%)	Total: 1152 (75.590553%)
[0.100102, 0.100482)	                    	Count: 2 (0.131234%)	Total: 1154 (75.721786%)
[0.100482, 0.100861)	               #####	Count: 367 (24.081366%)	Total: 1521 (99.803146%)
[0.100861, 0.101241)	                    	Count: 2 (0.131234%)	Total: 1523 (99.934380%)
[0.101241, 0.101620)	                    	Count: 0 (0.000000%)	Total: 1523 (99.934380%)
[0.101620, 0.102000]	                    	Count: 1 (0.065617%)	Total: 1524 (100.000000%)
```

looks mostly good.

I did not do any tuning for your setup, so tracking mostly failed and the map looks awful, but it SLAMs.

[image]

Some comments:

1. I don't know much about the size of your boat, but I assume that roll and pitch of the boat will be affected by the sea state. Since your setup has no IMU it is impossible for Cartographer to track that based on 2D LiDAR data alone which will strongly affect your matching performance.

2. The random LiDAR returns from the LiDAR hitting the water creating spurious reflections are also difficult. Is there a way to mount the LiDAR higher?

3. From what I can tell the LiDAR sees mostly undescript features from the cost line. Even as a human it is hard to a point on the cost in the noisy sensor readings.

elgarbe, 2018-03-26:
I will attach Hctor Slam result of that bag file later, meanwhile here you can find my boat in the mapping process.
https://youtu.be/Jmvn3rEICh4

and [here](https://discuss.ardupilot.org/t/leonardos-research-boat/27212) you can find some pictures of the boat
I think that laser beam don't hit the watter, but I do think that the coast is noisy. 

I will try to add IMU information. Could you share the config file used to produce your results, please?

Thank!

cschuet, 2018-03-26:
Branch with the modified configuration files is [here](https://github.com/cschuet/cartographer_ros/tree/boat).

elgarbe, 2018-03-27:
this is hector slam result for previous bag file.

[image]

From 0 to 50 sec there are 2 parts where slam fail a little. Then from 50 to 104 sec there are a really good slam. Is there a way to get better map with cartographer and this bag file?

I would like to start working with cartographer. So I would like to add IMU data, and maybe, odom data from pixhawk. Piwhawk provide IMU and other data through MavROS. I can go to the stream and record another bag file, with LIDAR and IMU data. But, do I need to run a tf node (tf2?) that transform IMU and LIDAR data into my boat frame? how does they be defined?

I need a starting point on this.
Thank!

ojura, 2018-03-27:
Have you seen the [tuning guide](http://google-cartographer-ros.readthedocs.io/en/latest/tuning.html)? Local SLAM, especially the ceres scan matcher (e.g. the occupancy grid, translation and rotation weights) can be a bit difficult to tune, you have to fiddle around with it until you get a feel for what works and what not.

elgarbe, 2018-03-27:
I need to get cartogrpaher working first, the I will try tunning guide.

Meanwhile, I would like to record a bag file with Laser and IMU data.
Ardurover publish IMU data into base_link frame through mavros. Do I need to run a TF node (TF2?) to transform laser into base_link frames? adding that TF is all I need to get cartographer working with IMU data?

Thank!

elgarbe, 2018-03-28:
I'm trying to get the same result as @cschuet so I run:
```
git remote add cschuet https://github.com/cschuet/cartographer_ros.git
git fetch cschuet
git checkout boat
```

Then run 
`roslaunch cartographer_ros backpack_2d.launch `

and get this

```
bag_filename:=${HOME}/19_03_18_01.bag
... logging to /home/elgarbe/.ros/log/b431fd54-32d8-11e8-a35c-7c67a237b388/roslaunch-pc-3717.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pc:33208/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [3729]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b431fd54-32d8-11e8-a35c-7c67a237b388
process[rosout-1]: started with pid [3742]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [3745]
process[cartographer_node-3]: started with pid [3746]
process[cartographer_occupancy_grid_node-4]: started with pid [3747]
[ INFO] [1522276699.985227242]: I0328 19:38:19.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1522276700.009821999]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522276700.010141578]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522276700.022544247]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522276700.022700134]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522276700.034212513]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522276700.034401673]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522276700.041112461]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522276700.041322069]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522276700.053985290]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522276700.054380598]: I0328 19:38:20.000000  3746 configuration_file_resolver.cc:41] Found '/home/elgarbe/catkin_ws_cart/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522276700.161090686]: I0328 19:38:20.000000  3746 submap_2d.cc:203] Added submap 1
[ INFO] [1522276700.176554106]: I0328 19:38:20.000000  3746 map_builder_bridge.cc:130] Added trajectory with ID '0'.
```
and nothing else, not rviz

elgarbe, 2018-03-28:
If I run 
`roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/19_03_18_01.bag`

then, get this:

[image]

elgarbe, 2018-03-29:
I don't understand why all of that frames is showing in rviz. my bag file has just scan topic on laser frame.
So @cschuet configure 
  tracking_frame = "laser",
  published_frame = "laser",
odom_frame = "laser",

I need to get this working to start testing tuning of my bags files...

elgarbe, 2018-03-29:
never mind, after go to cschuet branch I have to rebuild. Now is working.

elgarbe, 2018-03-29:
Processing another bag file:
https://youtu.be/o-9h0f7Ot3g

closed 2018-03-29 by elgarbe

Abhirup0209, 2023-03-19:
How did you rebuild your cartographer workspace?