cartographer-project/cartographer_ros #784: I can't re build catkin_ws

elgarbe, 2018-03-27:
I made some modification to lua and launch files and want to build ws again.
I run:

catkin_make_isolated --install --use-ninja

and a I get:

```
==> Processing catkin package: 'cartographer_rviz'
==> Building with env: '/home/elgarbe/catkin_ws_cart/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz'
[1/1] Re-running CMake...
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /home/elgarbe/catkin_ws_cart/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   iostreams
--   regex
-- Using CATKIN_DEVEL_PREFIX: /home/elgarbe/catkin_ws_cart/devel_isolated/cartographer_rviz
-- Using CMAKE_PREFIX_PATH: /home/elgarbe/catkin_ws_cart/devel_isolated/cartographer_rviz;/home/elgarbe/catkin_ws_cart/install_isolated;/home/elgarbe/catkin_ws/devel;/opt/ros/kinetic;/home/elgarbe/catkin_ws_test/devel
-- This workspace overlays: /home/elgarbe/catkin_ws_cart/devel_isolated/cartographer_rviz;/home/elgarbe/catkin_ws_cart/install_isolated;/home/elgarbe/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz/test_results
-- Found gmock: gmock and gtests will be built
CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:381 (add_library):
  add_library cannot create imported target "gmock" because another target
  with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:35 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:383 (add_library):
  add_library cannot create imported target "gmock_main" because another
  target with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:35 (find_package)

-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.11
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using Qt5 based on the rviz_QT_VERSION: 5.5.1
-- Configuring incomplete, errors occurred!
See also "/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz/CMakeFiles/CMakeOutput.log".
See also "/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz/CMakeFiles/CMakeError.log".
FAILED: /usr/bin/cmake -H/home/elgarbe/catkin_ws_cart/src/cartographer_ros/cartographer_rviz -B/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz
ninja: error: rebuilding 'build.ninja': subcommand failed
<== Failed to process package 'cartographer_rviz': 
  Command '['/home/elgarbe/catkin_ws_cart/install_isolated/env.sh', 'ninja', 'build.ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/elgarbe/catkin_ws_cart/build_isolated/cartographer_rviz && /home/elgarbe/catkin_ws_cart/install_isolated/env.sh ninja build.ninja

Command failed, exiting.
```

cschuet, 2018-03-27:
Seems this is related to a recent gmock fix. @gaschler FYI. Can you

rm -rf *isolated

and try again?

elgarbe, 2018-03-27:
Ok, I will try it when I return to home.
If a made a modification to a launch file or to a lua file, do I need to rebuild cartographer?
Do I need to run 
catkin_make_isolated --install --use-ninja
or just
catkin_make_isolated

Thank!

ojura, 2018-03-27:
No need to rebuild. Also make sure you pull the newest googlecartographer/cartographer repo.

elgarbe, 2018-03-28:
runinng rm -rf *isolated and running catkin_make_isolated again do not work:

`CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:381 (add_library):
  add_library cannot create imported target "gmock" because another target
  with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:45 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:383 (add_library):
  add_library cannot create imported target "gmock_main" because another
  target with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:45 (find_package)

-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.11
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found Lua: /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.2.4") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_common.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_common.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   iostreams
--   regex
-- Configuring incomplete, errors occurred!
See also "/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/elgarbe/catkin_ws_cart/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/elgarbe/catkin_ws_cart/install_isolated/env.sh', 'cmake', '/home/elgarbe/catkin_ws_cart/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/elgarbe/catkin_ws_cart/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/elgarbe/catkin_ws_cart/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/elgarbe/catkin_ws_cart/build_isolated/cartographer_ros && /home/elgarbe/catkin_ws_cart/install_isolated/env.sh cmake /home/elgarbe/catkin_ws_cart/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/elgarbe/catkin_ws_cart/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/elgarbe/catkin_ws_cart/install_isolated -G Ninja

Command failed, exiting.
`
I need instructions about how to pull newest code.

ojura, 2018-03-28:
Run `git pull`. Or just `git clone` everything again.

elgarbe, 2018-03-28:
Ok, now is working, thank!

closed 2018-03-28 by elgarbe

LOKAKSHAI, 2021-06-08:
Failed to process package 'cartographer_ros'

MakeFiles/Makefile2:3315: recipe for target 'CMakeFiles/cartographer_ros.dir/all' failed
make[1]: *** [CMakeFiles/cartographer_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'cartographer_ros': 
  Command '['/home/catkin_ws/devel_isolated/tf2_ros/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/catkin_ws/build_isolated/cartographer_ros && /home/catkin_ws/devel_isolated/tf2_ros/env.sh make -j8 -l8