cartographer-project/cartographer_ros #786: tf between map and odom not being published - 2D cartographer, car-like robot, 2 LIDARs and odometry

davidromanos, 2018-03-27:
Hi everyone, I am trying to map a small laboratory with a car-like robot (a golfcart). The robot is equipped with two LIDARs, one facing forward placed at wheel's center level and another one placed at  roof level. This second LIDAR is placed with a roll angle of pi rad and it is tilted with a pitch angle of 0.31 rad.

The robot is also equipped with encoders on the rear wheels and an encoder on the steering column. I use this to publish the odometry, obtaining quite good results.

Now I would like to run cartographer_ros with odometry to build the occupancy map of the laboratory, but I can't manage to configure it to work. So far I don't get any error nor warning on the terminal but the tf between map --> odom is not being published.

Please, find here my files:
-  [Configuration file](https://www.dropbox.com/s/ecpmrp61jym7clx/aaucart_2d.lua?dl=0)
- [launch file](https://www.dropbox.com/s/k6ekv16s3wis7px/aaucart_2d.launch?dl=0)
- [urdf](https://www.dropbox.com/s/yagmgtkgb3p9oqe/aaucart_2d.urdf?dl=0)
- [bagfile](https://www.dropbox.com/s/tg8xzrr723i1sch/aaucart.bag?dl=0)
- The output of running the rosbag_validate is [here](https://gist.github.com/davidromanos/cb32cc5df1e4eabf6b4ef29eac0656f7).

I hope you can help me, thanks in advance.

MichaelGrupp, 2018-03-28:
Have a look at the meaning of the frame parameters: https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html

In your case, I'd try:
``` 
map_frame = "map",
tracking_frame = "base_link",
published_frame = "base_link",
odom_frame = "odom",
provide_odom_frame = true,
```

davidromanos, 2018-03-28:
Thanks for your quick reply @MichaelGrupp! I am new at cartographer_ros, but following the configuration instructions I would say that the configuration that you suggest will publish odometry, am I right? What I am interested is rather the opposite, I want to use a separate odometry source, recorded on the rosbag as /odom, which is based on wheel encoders. Following the configuration instructions I set the configuration to:

```
map_frame = "map",
tracking_frame = "base_link",
published_frame = "odom",
odom_frame = "odom",
provide_odom_frame = false,
publish_frame_projected_to_2d = false,
use_odometry = true,
```
If I do as you suggest, `/cartographer_node` doesn't subscribe to `/odom`, to do so I understand I have to set `use_odometry = true` but do I have to set `provide_odom_frame = true` as well? The rosbag that I provide contains topic `/odom` with `frame_id="odom"` and `child_frame_id="base_link"`, as well as `/tf ` topic publishing the transformation from odom-->base_link.

I followed the configuration instructions but I cannot manage to make it work, can you see what am I doing wrong? Thanks again!

MichaelGrupp, 2018-03-28:
Thanks for the info, I have to admit I didn't try your bagfile because I don't really have time at the moment and my suggestion was a guess.

The term "odom" is always confusing in the ROS world because it is the common name for both a `nav_msgs/Odometry` topic and a TF frame.

* `use_odometry = true` means that you subscribe to the `nav_msgs/Odometry` topic as an input for sensor fusion in local SLAM (what you want).
* `provide_odom_frame = false` is ok in your case if you say that the odom frame is already published by another source.

It looks like the frame settings look reasonable for what you want to achieve. Are you sure there's no information in the logs? Could you upload the output of one of your test runs? Usually Cartographer outputs useful information if something goes wrong.

davidromanos, 2018-03-29:
Thanks again @MichaelGrupp. Here the output of my test run:

```
... logging to /home/aaucart/.ros/log/5ebb7d4a-3324-11e8-93b1-a088b4da265c/roslaunch-aaucart-Latitude-E6520-12643.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://aaucart-Latitude-E6520:43795/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: lunar
 * /rosversion: 1.13.6
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [12653]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5ebb7d4a-3324-11e8-93b1-a088b4da265c
process[rosout-1]: started with pid [12666]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [12669]
process[cartographer_node-3]: started with pid [12682]
process[cartographer_occupancy_grid_node-4]: started with pid [12689]
process[rviz-5]: started with pid [12719]
process[playbag-6]: started with pid [12733]
[ INFO] [1522309194.190123031]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/aaucart_2d.lua' for 'aaucart_2d.lua'.
[ INFO] [1522309194.190702357]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522309194.190806651]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1522309194.190946263]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522309194.191038304]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1522309194.191323258]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522309194.191428210]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1522309194.191564651]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522309194.191658757]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1522309194.191855954]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522309194.191957459]: I0329 09:39:54.000000 12682 configuration_file_resolver.cc:41] Found '/home/aaucart/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1522309194.207402400]: I0329 09:39:54.000000 12682 submap_2d.cc:203] Added submap 1
[ INFO] [1522309194.207618215]: I0329 09:39:54.000000 12682 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1522309195.397894921, 1522139403.409500695]: I0329 09:39:55.000000 12682 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636577362032955239'.
```
And after some time running it starts to spit this warning repeatedly:

`[ WARN] [1522312960.369508891, 1522139589.592720460]: W0329 10:42:40.000000 13903 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan_1)`

Also, please take a look at the tf tree [here](https://www.dropbox.com/s/qn86oclxdemeis2/frames.pdf?dl=0) yielded by running `rosrun tf2_tools view_frames.py`.

davidromanos, 2018-04-03:
Hi, at the end resulted to be a problem of the rosbag. I still don't understand why, but just recorded a new bag and seems to work fine. Still mapping results are quite awful but I guess it needs some tunning.
Thanks for your time @MichaelGrupp!

closed 2018-04-03 by davidromanos

MichaelGrupp, 2018-04-03:
Ok! :) Usually something is wrong outside of Cartographer with the "waiting for data" message, or a topic is not remapped correctly etc.

I'd recommend looking at the tuning guide in the official cartographer_ros documentation page.

Mafuqiang, 2019-05-29:
can you run cartographer with odometry? I try it ,but there are some mistake in my configfile.