cartographer-project/cartographer_ros #792: I had problems when I used odometry in cartographer

Vincent1923, 2018-03-31:
Dear cartographer ros team,
Thank you for your maintenance of this open source SLAM. I've been using google cartographer for a couple of months, but I have encountered some problems recently. The result was good when I only used the sensor_msgs/LaserScan in cartographer. But when I tried to add the odometry to achieve better results, the map was bad. The commandline I used is `roslaunch cartographer_ros demo_hokuyo_2d_odom.launch bag_filename:=${HOME}/Downloads/2018-03-21-15-52-22.bag`. After the map had been built for a while, the trajectory of the robot would have a big slide: 
[image]

I wonder if this is the effect of the frequency of the LaserScan and Odometry. Their frequency is as follows.
[image]
After I reduced the “odometry_sampling_ratio” from 1 to 0.15, the slip of the trajectory no longer appeared. I don't know if it is correct when I modified this parameter.

Here is my files:

- [Theoutput of the bag file validate tool.](https://gist.github.com/Vincent1923/216b2266521678703827443e95f6d961)

- [All the configuration, launch, and URDF files are here.](https://github.com/Vincent1923/cartographer_ros)

- [And this is the bag file I used.](https://drive.google.com/open?id=12s4Av5XlsxsTWT43Iy3QEUAF7Cbu8nxL)

Any help would be greatly appreciated.

gaschler, 2018-04-03:
> I0330 16:19:39.266774 113606 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
> Count: 127911  Min: 0.000002  Max: 0.122347  Mean: 0.019991
> [0.000002, 0.012237)	    ################	Count: 102329 (80.000153%)	Total: 102329 (80.000153%)
> [0.073409, 0.085644)	                    	Count: 0 (0.000000%)	Total: 102329 (80.000153%)
> [0.085644, 0.097878)	                    	Count: 2665 (2.083480%)	Total: 104994 (82.083633%)
> [0.097878, 0.110113)	                ####	Count: 22849 (17.863201%)	Total: 127843 (99.946838%)

It's strange that your minimum time difference between odometry messages is 0. 
Just a wild guess, are there duplicate messages?

Please inspect the details of the odometry messages with `rqt_bag`.

> After I reduced the “odometry_sampling_ratio” from 1 to 0.15,

This throws away all but 15% of messages, it only covers the real issue.

Vincent1923, 2018-04-11:
Thank you for your answers! 
I have checked the odometry messages with rqt_bag. You are right about the odometry messages. There are indeed a lot of duplicate odometry messages in my data. Here is the output of the rqt_bag.
[image]
And the time interval for every two odometry messages is as follows.
[image]
Is this the reason why the trajectory of the robot has a big slide as the picture shows above?

Since I knew there may be some problems with the odometry messages, I have made changes to the odometry messages. Here is the time interval for every two odometry messages after I made the changes.
[image]
However, there were still some problems when I added the odometry messages in cartographer. The result is not good. Here is the map when I used the changed odometry messages.
[image]
But if I didn't use the odometry messages, the result was better as follows.
[image]
I really feel confused about this. Could you please help me? Are the odometry messages still wrong in my data?
Thanks you very much!