cartographer-project/cartographer_ros #794: Problems adding odometry

elgarbe, 2018-04-01:
I was able to add IMU data and really improve SLAM process.
Now I want to add odometry msg.
Could someone explain what this errors and warnings mean:

```
rosrun cartographer_ros cartographer_rosbag_validate -bag_filename /home/elgarbe/casa_imu_odom.bag 
E0331 22:00:08.037511 21914 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan has serialization time 1522539150.304405142 but sensor time 1522539149.638893094 differing by -0.665512 s.
E0331 22:00:08.037817 21914 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan has serialization time 1522539150.304603005 but sensor time 1522539149.739401604 differing by -0.565201 s.
W0331 22:00:08.037866 21914 rosbag_validate_main.cc:344] frame_id "laser" on topic /scan has serialization time 1522539150.304968991 but sensor time 1522539149.838816476 differing by -0.466153 s.
E0331 22:00:08.038041 21914 rosbag_validate_main.cc:318] frame_id "base_link" is send on multiple topics. It was seen at least on /mavros/imu/data and /mavros/imu/data_raw
E0331 22:00:08.038115 21914 rosbag_validate_main.cc:318] frame_id "map" is send on multiple topics. It was seen at least on /mavros/global_position/local and /mavros/local_position/odom
E0331 22:00:08.038166 21914 rosbag_validate_main.cc:318] frame_id "map" is send on multiple topics. It was seen at least on /mavros/global_position/local and /mavros/local_position/odom
E0331 22:00:08.097421 21914 rosbag_validate_main.cc:381] IMU data (frame_id: "base_link") has a large gap, largest is 0.220583 s, recommended is [0.0005, 0.005] s with no jitter.
I0331 22:00:08.097544 21914 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/mavros/imu/data" (frame_id: "base_link"):
Count: 1160  Min: 0.019858  Max: 0.220583  Mean: 0.119986
[0.019858, 0.039931)	          ##########	Count: 580 (50.000000%)	Total: 580 (50.000000%)
[0.039931, 0.060003)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.060003, 0.080076)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.080076, 0.100148)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.100148, 0.120221)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.120221, 0.140293)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.140293, 0.160366)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.160366, 0.180438)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.180438, 0.200511)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
[0.200511, 0.220583]	          ##########	Count: 580 (50.000000%)	Total: 1160 (100.000000%)
E0331 22:00:08.097582 21914 rosbag_validate_main.cc:374] Point data (frame_id: "laser") has a large gap, largest is 0.103742 s, recommended is [0.0005, 0.05] s with no jitter.
I0331 22:00:08.097654 21914 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser"):
Count: 1399  Min: 0.097248  Max: 0.103742  Mean: 0.099966
[0.097248, 0.097897)	                    	Count: 1 (0.071480%)	Total: 1 (0.071480%)
[0.097897, 0.098547)	                    	Count: 2 (0.142959%)	Total: 3 (0.214439%)
[0.098547, 0.099196)	                    	Count: 30 (2.144389%)	Total: 33 (2.358828%)
[0.099196, 0.099846)	            ########	Count: 572 (40.886349%)	Total: 605 (43.245174%)
[0.099846, 0.100495)	                ####	Count: 293 (20.943531%)	Total: 898 (64.188705%)
[0.100495, 0.101144)	             #######	Count: 496 (35.453896%)	Total: 1394 (99.642601%)
[0.101144, 0.101794)	                    	Count: 4 (0.285919%)	Total: 1398 (99.928520%)
[0.101794, 0.102443)	                    	Count: 0 (0.000000%)	Total: 1398 (99.928520%)
[0.102443, 0.103092)	                    	Count: 0 (0.000000%)	Total: 1398 (99.928520%)
[0.103092, 0.103742]	                    	Count: 1 (0.071480%)	Total: 1399 (100.000000%)
I0331 22:00:08.097743 21914 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/mavros/global_position/local" (frame_id: "map"):
Count: 2783  Min: 0.000199  Max: 0.104984  Mean: 0.050013
[0.000199, 0.010678)	          ##########	Count: 1391 (49.982033%)	Total: 1391 (49.982033%)
[0.010678, 0.021156)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.021156, 0.031635)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.031635, 0.042113)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.042113, 0.052592)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.052592, 0.063070)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.063070, 0.073549)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.073549, 0.084027)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.084027, 0.094506)	                    	Count: 0 (0.000000%)	Total: 1391 (49.982033%)
[0.094506, 0.104984]	          ##########	Count: 1392 (50.017967%)	Total: 2783 (100.000000%)

```

there is /mavros/global_position/local and there is not the /mavros/local_position/odom topic, why?

my config files are here:
https://github.com/elgarbe/cartographer_ros/tree/odom_imu/cartographer_ros

what does TRAJECTORY_BUILDER_2D.num_accumulated_range_data meaning?

thank

thank!

gaschler, 2018-04-03:
> E0331 22:00:08.038041 21914 rosbag_validate_main.cc:318] frame_id "base_link" is send on multiple topics. It was seen at least on /mavros/imu/data and /mavros/imu/data_raw

In this case this is fine because it is in the namespace.

> E0331 22:00:08.097421 21914 rosbag_validate_main.cc:381] IMU data (frame_id: "base_link") has a large gap, largest is 0.220583 s, recommended is [0.0005, 0.005] s with no jitter.

Should be fine.

> E0331 22:00:08.037511 21914 rosbag_validate_main.cc:342] frame_id "laser" on topic /scan has serialization time 1522539150.304405142 but sensor time 1522539149.638893094 differing by -0.665512 s.

Marginally ok. Is your machine slow and cannot keep up with serialization?

> I0331 22:00:08.097544 21914 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/mavros/imu/data" (frame_id: "base_link"):
> Count: 1160  Min: 0.019858  Max: 0.220583  Mean: 0.119986
> [0.019858, 0.039931)	          ##########	Count: 580 (50.000000%)	Total: 580 (50.000000%)
> [0.180438, 0.200511)	                    	Count: 0 (0.000000%)	Total: 580 (50.000000%)
> [0.200511, 0.220583]	          ##########	Count: 580 (50.000000%)	Total: 1160 (100.000000%)

Strangely high jitter. Are you sending messages twice?

> TRAJECTORY_BUILDER_2D.num_accumulated_range_data

Number of TimedPointCloudData structures that are one laser scanner revolution and should be merged before scan matching.
If subdivisions are used, usually the number of range sensors times number of subdivisions.
Confirm correct setting by looking at the scan visualization in rviz, it should cover 360 degrees and not flicker too much.

elgarbe, 2018-04-03:
The computer that records laser and IMU data is a Raspberry Pi 3. It receive IMU data through mavros, using uart port at 921.600bps (there are several msgs). I'm not sure if it has enough processing power, but I think that it has.
I found that sometimes IMU data has that stage jitter, I will investigate what happen there.
Tomorrow I will run some test on the stern.

Thank!