cartographer-project/cartographer_ros #795: which part of odom msg cartographer use?

elgarbe, 2018-04-01:
this is my odom mesage:

```
header: 
  seq: 1556
  stamp: 
    secs: 1455208296
    nsecs: 310980854
  frame_id: "map"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -0.005535364151
      y: -0.267928063869
      z: 0.0810627788305
    orientation: 
      x: -0.0192797667306
      y: -0.0368807024663
      z: 0.50707105612
      w: -0.860898976899
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
twist: 
  twist: 
    linear: 
      x: -0.00152582221037
      y: -0.0375590241358
      z: -0.00955859516847
    angular: 
      x: -7.68475001678e-06
      y: -0.000235044397414
      z: 0.00026510679163
  covariance: [0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001]
---
```

Wich part of this msg is used by cartographer?

ojura, 2018-04-01:
Only the pose.

elgarbe, 2018-04-01:
My odom msg provider is a pixhawk running ardurover. I think tha pose is the result of an EKF of GPS and IMU. My boat don't have wheel encoder, of course.
Is there a way to check how well perform odom msg? I'm thinking on some ros package (I don't know which one) that allows me to plot pose msg and visually compare with the path I follow. Do you know any?
I've allready try to add odometry to cartographer and it performs worse...
Can you give a look at this:
https://github.com/googlecartographer/cartographer_ros/issues/794
please?

Thank you very much!

BTW: Here it is a basic comparison of cartographer with and without IMU. 
https://www.youtube.com/watch?v=qidhtue8xag

ojura, 2018-04-02:
No idea how to check odometry quality, but you might be better off using GPS directly, cartographer_ros can take in `sensor_msgs::NavSatFix` messages directly.

gaschler, 2018-04-03:
> you might be better off using GPS directly 

Agree.

>  I think that pose is the result of an EKF of GPS and IMU.

This is a very different noise model from "odometry". Generally, it may be worth passing the GPS data alone as NavSatFix for comparison.

elgarbe, 2018-04-03:
But there is no NavSatFix implementation for 2D. I already try that.